evil-mad/EggBot #23: Update instructions for installation

secretgspot, 2015-01-28:
Current folder structure makes no sense and full of stuff that has nothing to do with instructions on wiki, how does one supposed to install the drivers and inkscape extension?  I've read and re-read instructions 100 times, still can't install the driver, from where, which file, which folder?

oskay, 2015-01-28:
The instructions both on the wiki and on the releases page are much simpler than you are implying. 

They say to (1) download and install Inkscape, and (2) then to download the installer, and to run the installer. As noted, the windows driver (not required for Mac/Linux) is now included with the installer. 

Are you really having that much trouble understanding it, after reading it 100 times?  If so, please clarify exactly problem you are having.

secretgspot, 2015-01-28:
did find release page after posting this.  go ahead and close this :D

closed 2015-01-28 by secretgspot

oskay, 2015-01-28:
Great!