evil-mad/EggBot #25: Add mphidflash (or replacement) to this repository.

oskay, 2015-04-11:
As the old repository is shutting down, we should get the EBB flashing tool moved over here, or perhaps to its own repository. Because of the way that our "releases" pages are structured, it would be nice if it didn't have to be there in that list of downloads.

Also, it has gotten harder to run on Ubuntu lately...
https://code.google.com/p/eggbotcode/issues/detail?id=38

@EmbeddedMan: OK to assign this to you?

EmbeddedMan, 2015-04-11:
Yup, assign it to me. I'll take care of it this weekend. I'm not much of a
Linux user, so I'm not sure how far I can get with that issue. But I'll
give it a go.

*Brian

EmbeddedMan, 2015-04-12:
OK, I've migrated mphidflash to GitHub. There is a 1.6 release zip file that contains binaries for each supported platform that you could link to if you want : (https://github.com/EmbeddedMan/mphidflash/blob/master/dist/mphidflash-1.6-bin.zip)

I'm now working on trying to get an Ubuntu 10.10 system running so I can test out the problems there.

EmbeddedMan, 2015-04-12:
OK, is it critical that we support Ubuntu 10.10? (This release was EOLed in 2012, and it is really difficult to do much with it because the normal commands to update your system, etc. don't work. For example you can't download Inkscape for it because it's too old, etc.)

If you want me to dive into this some more, I can, but so far I've spent several hours on bringing up a 10.10 system, and realize I'm way over my head. 

*Brian

oskay, 2015-04-12:
Linux "in general" should be supported, but it is not necessary to support EOL Ubuntu versions. You should note that one version of Ubuntu 10 (Ubuntu 10.04.4 LTS Server) is technically supported until April 30.

oskay, 2015-04-12:
I'll mark this issue closed, since the root issue has been resolved. Feel free to loop me in on any issues that need my attention. :D

closed 2015-04-12 by oskay

EmbeddedMan, 2015-04-12:
Ahh, I see. I'll see if I can have an easier time testing mphidflahs out on 10.04.4. I'll let you know. It would be really nice if we could say "Yup, this works".

oskay, 2015-04-12:
It may be easier to just update to the current version (e.g., 14.04).

EmbeddedMan, 2015-04-12:
I now have a VM with Ubuntu 14.04. I have installed Inkscape, and have verified normal EB functionality. I have also confirmed that mphidflash 1.6 fails to detect an EBB that is in bootloader
mode, and so can not update the firmware on the board, although the error is different from the one reported: lib hid-dev  not installed by default

Instead, I get "Warning: cannot claim interface: could not claim interface 8: Operation not permitted"

I will continue to dig into this problem.

oskay, 2015-04-12:
Sounds like a permissions error. Retry with sudo?

oskay, 2015-04-12:
Possibly related to the dialout group permissions thing?

EmbeddedMan, 2015-04-12:
Yup, you were right. Running mphidflash with sudo worked. I had no trouble
updating the EBB with a new firmware version. (Previously I couldn't get
Inkscape to see the EBB until I set myself as part of the dialout group -
just like your Wiki says - and that allowed normal EB operation. But this
mphidflash thing was different.)

So, what to do next - mphidflahs works fine (as far as I can tell) on
Ubuntu 14.04. But possibly has problems with 10.10. It feels to me like
somebody who knows a lot more about Linux and USB will be needed to solve
the 10.10 problem.

oskay, 2015-04-13:
I'd suggest including sudo in your command line example, then. 

10.10 specifically is likely not worth the effort at this point _unless_ someone comes to us needing help with it.

EmbeddedMan, 2015-04-13:
Good idea. The readme.txt has been updated to include this information now.