evil-mad/EggBot #27: Upper/lower limit of step moves seems to be +/- 65535

bmaranville, 2015-08-14:
From the documentation, for firmware 2.2.7 the number of steps should be a 32-bit signed integer.  It seems to only accept a 16-bit signed integer.  Output from my test program (note that KeyboardInterrupt triggers an EStop):

In [4]: ic.mc.sendCMD('SM,10000,0,-65535')
ic.mc.sendCMD('SM,10000,0,-65535')
Out[4]: 'OK'

In [5]: ic.mc.sendCMD('SM,10000,0,-65536')
ic.mc.sendCMD('SM,10000,0,-65536')
Out[5]: 'Major malfunction Axis2 StepCounter too high : 420906'
...
In [7]: ic.mc.sendCMD('SM,10000,0,65536')
ic.mc.sendCMD('SM,10000,0,65536')
Out[7]: 'Major malfunction Axis2 StepCounter too high : 420906'

In [8]: 
KeyboardInterrupt

In [8]: ic.mc.sendCMD('SM,10000,0,65535')
ic.mc.sendCMD('SM,10000,0,65535')
Out[8]: 'OK'

EmbeddedMan, 2015-08-14:
Yes, this is a bug that was discovered by another user, and I haven't had
time to fix it yet. Are you badly in need of 24 bit step values? (I think I
might not be able to go up to 32 bit for these, but may need to limit it to
3 bytes.)

bmaranville, 2015-08-14:
It would help us a lot to have 24 bit values... we are driving a goniometer with (1 stepper motor revolution) = 1 degree change of goniometer.  We will frequently want to drive the goniometer up to 180 degrees or 360 degrees, which would be 1152000 steps if we are using 1/16 microstepping.  With 24 bits we would get more than 8x360 degrees and that would be more than enough

EmbeddedMan, 2015-08-14:
OK, I'll see if I can get the full 3-byte values working for you shortly.

EmbeddedMan, 2015-08-16:
I _think_ I found the bug that was preventing your 3-byte step values from working. Please check out the new version 2.2.8, and see if you can find any problems with 3-byte durations or 3-byte step values. I did some simple tests here, and they all pass. 

https://github.com/evil-mad/EggBot/blob/master/EBB_firmware/Releases/app/EBF_v228.zip?raw=true

Let me know if this works and I'll close this issue out. Or if you can find more cases where it doesn't work, then I can work on fixing those.

bmaranville, 2015-08-17:
I just did some moves of +/- 1152000 steps, and they worked just great.

EmbeddedMan, 2015-08-17:
That's fantastic. You can now do single moves that are up to 4.6 hours in
length. Hopefully that will be enough. :-)

I'll close this issue, and you can open another one if you run into any
other issues.

*Brian

closed 2015-08-17 by EmbeddedMan