evil-mad/EggBot #28: Return steps moved so far when aborting move

bmaranville, 2015-08-18:
This would be an enhancement to the brand-new much-appreciated EStop command: 
(I don't have a working compiler for the firmware, so I can't test this, so it's not a proper pull request)

I wonder if this patch would make it possible to get back the number of steps actually moved before abort: 

```
diff --git a/EBB_firmware/app.X/source/ebb.c b/EBB_firmware/app.X/source/ebb.c
index 1daa381..37ba6a0 100644
--- a/EBB_firmware/app.X/source/ebb.c
+++ b/EBB_firmware/app.X/source/ebb.c
@@ -1050,18 +1050,20 @@ static void process_SM(
 
 // E-Stop
 // Usage: ES<CR>
-// Returns: 0 if SM command was not interrupted, 1 if it was, then OK<CR>
+// Returns: <command_interrupted>,<steps_achieved1>,<steps_achieved2><CR>OK<CR>
 // This command will abort any in-progress motor move (SM) command.
 // It will also clear out any pending command(s) in the FIFO.
-// It will return with a 1 if an SM command was aborted or if an SM command 
-// was waiting in the FIFO and had to be discarded. This 1 indicates that some
-// position information may be lost, since one or more SM commands was not
-// finished.
-// It will return a 0 if no SM command was executing at the time, and no SM
-// command was 
+// It will return with a 1 for command_interrupted if an SM command was aborted 
+// or if an SM command was waiting in the FIFO and had to be discarded. 
+// The next return values steps_achieved1 and steps_achieved2 are then the number 
+// of steps that were moved on axes 1 and 2 before the abort command was sent.
+// It will return 0,0,0 if no SM command was executing at the time, and no SM
+// command was in the FIFO.
 void parse_ES_packet(void)
 {
-    UINT8 retval = 0;
+    UINT8 retval1 = 0;
+    UINT32 retval2 = 0;
+    UINT32 retval3 = 0;
 
     // If there is a command waiting in the FIFO and it is a move command
     // or the current command is a move command, then remember that for later.
@@ -1071,7 +1073,7 @@ void parse_ES_packet(void)
         CurrentCommand.Command == COMMAND_MOTOR_MOVE
     )
     {
-        retval = 1;
+        retval1 = 1;
     }
     
     // If the FIFO has a move command in it, remove it.
@@ -1087,11 +1089,13 @@ void parse_ES_packet(void)
     if (CurrentCommand.Command == COMMAND_MOTOR_MOVE)
     {
        CurrentCommand.Command = COMMAND_NONE;
+       retval2 = CurrentCommand.StepsCounter[0];
+       retval3 = CurrentCommand.StepsCounter[1];
         CurrentCommand.StepsCounter[0] = 0;
         CurrentCommand.StepsCounter[1] = 0;
     }
                 
-    printf((far rom char *)"%d\n\r", retval);
+    printf((far rom char *)"%d,%d,%d\n\r", retval1, retval2, retval3);
        print_ack();
 }
```

EmbeddedMan, 2015-08-18:
Heck yeah. That's a great idea. I'll merge that in tonight and test for a
new release.

*Brian

EmbeddedMan, 2015-08-18:
OK, this got a bit more involved than your patch suggests, but it's all OK now. 

There is a new release 2.2.9 (https://github.com/evil-mad/EggBot/raw/master/EBB_firmware/Releases/app/EBF_v229/EBV_v229.hex) that I'd like you to try. The new format of the ES command is documented in the EBB commands page (http://www.schmalzhaus.com/EBB/EBBCommands.html)

Please give it a try and see if it meets your needs.

oskay, 2018-07-08:
Closing this due to apparent success and lack of feedback from @bmaranville. Feel free to open if necessary.

closed 2018-07-08 by oskay