evil-mad/EggBot #29: Add new "XM" command for mixed-axis stepper moves

oskay, 2015-08-28:
Feature request:  Add a new command, that simplifies use of the EBB in machines that have mixed-axis geometry, such as [H-Bot](http://www.multimechatronics.com/build_a_h-bot.php) and [CoreXY](http://corexy.com).  

In machines with these and similar geometries, the rotation of a single stepper motor causes a translation of the machine's end effector in a direction _diagonal_ to the machine's cartesian axes &mdash; i.e., in the &pm; (X + Y) or &pm; (X - Y) directions. This is a well-understood type system, and is usable as-is with the EBB. However there are certain circumstances, such as during initial setup and test, where it is convenient to be able to issue an EBB command to move the machine directly alone its cartesian axes.

I would propose to add the following command to the [EBB command set](http://www.schmalzhaus.com/EBB/EBBCommands.html): 
`"XM,<move_duration>,<axisA>,<axisB><CR>"`

Executing this command would be equivalent to executing:
`"SM,<move_duration>,<axis1>,<axis2><CR>"`, 
where `<axis1> = <axisA> + <axisB> ` and `<axis2> = <axisA> - <axisB> `

@EmbeddedMan, I have assumed that this is _straightforward_, and could be done by (_insert hand-waving here_) having the XM command internally call the SM command. If it is not straightforward &mdash; for any reason &mdash; please say so. :)

EmbeddedMan, 2015-08-28:
> direction *diagonal* to the machine's cartesian axes — i.e., in the ± (X
> + Y) or ± (X - Y) directions. This is a well-understood type system, and is
> straightforward — for any reason — please say so. :)

Windell - that seems very straightforward. I'll try to get it in this
weekend. Do you have a way of immediately testing that the new command is
working properly?

*Brian

oskay, 2015-08-28:
We can certainly test it on a WaterColorBot, for example.

EmbeddedMan, 2015-08-28:
OK, new version of EBB Firmware is now up:
https://github.com/evil-mad/EggBot/blob/master/EBB_firmware/Releases/app/EBF_v230.zip?raw=true

Give it a try, and see if the XM command works as you intend it. If so, let
me know, and I'll close the issue.

*Brian

oskay, 2015-08-31:
I've uploaded this firmware and tested it. So far as I can tell, everything is working correctly. The scaling is a bit counterintuitive on a traditional cartesian machine, but it's certainly good enough to go.  

I've noted that support for CoreXY (and similar geometries) has been a factor for people to choose between different controller boards-- so this is a very nice feature to have!

EmbeddedMan, 2015-08-31:
Excellent!

closed 2015-08-31 by EmbeddedMan