evil-mad/EggBot #32: Bad hatch fill when user selects non-zero gap length

ShelMi, 2015-12-30:
When the raster traverses an object which is smaller than the selected gap length, the odd/even indicator for inside/outside of polygon gets reversed.

ShelMi, 2015-12-30:
I have a proposed fix, and it tests OK, but I am not up-to-speed on using git - and don't want to screw things up.

ShelMi, 2015-12-30:
Whoops - proves I don't know about git, I didn't mean to close the issue.

oskay, 2015-12-30:
Can you please state the issue that you are referring to very clearly, including how to reproduce it?

ShelMi, 2015-12-30:
Hi Windell...
        
            Wow - didn't realize you were a
          night-owl like me!  8^)  (I'm in
            Eastern time zone.)
            
                I
                have a very simple test image which shows the problem,
                the .svg file is attached.  Needless to say this is just
                a test image...I ran into the
                  problem with a rather more complicated image, which I
                  can supply if you want.
                
        Here is the Inkscape result
                                  image, with
                                    the red lines being strokes of my
                                    original svg rectangles of different
                                    sizes, and black being the output of
                                    Extensions/Eggbot Contributed/Hatch
                                    Fill... with 
                                            Hatch
                                    angle 90.0;
                                            Hatch
                                    spacing 6.0;
                                            Minimum
                                    hatch and gap length  4.0;
                                            Tolerance
                                    20.0;
                                              unchecked
                                      Crosshatch:
                                        
                                            I
                                          believe I have solved the
                                          problem, and can give further
                                          details if you wish.
                                          
                                             
                                            There is an additional
                                            problem with this extension,
                                            where
                                              though I haven't cured the
                                              problem, I
                                              have markedly improved the
                                              performance.  It's
                                                a little harder to
                                                document
                                          this one, so I'll just stick
                                            with this one - until you
                                            are convinced that I'm a
                                            reliable reporter.  8^)
                                          
                                          Bests,
                                              ...Shel
                                                Michaels
                                                NH,
                                                    USA
                                                
                
    On 2015-12-30 3:33 AM, Windell Oskay
      wrote:
    
    
      Can you please state the issue that you are referring to very
        clearly, including how to reproduce it?
      —
          it on GitHub.

ShelMi, 2015-12-31:
Maybe a description of the problem from the user's standpoint:  When doing a Hatch Fill from the contributed extensions in Inkscape, the hatch may have areas where it fills the space _between_ edges, where there is supposed to be no fill.  The result is hatch where you don't want it, and no hatch where you do want it.

This only happens when you specify a Minimum Hatch and Gap Length greater than zero.

The change I propose is in the insterstices procedure, in the "mind the gap" logic.

I propose changing the following snippet from this:

    # Remember the relative start and end of this hatch segment
        last_sa = [ sa[i], sa[i+1] ]
        i = i + 2
    else:
        i = i + 1

to this:

    # Remember the relative start and end of this hatch segment
        last_sa = [ sa[i], sa[i+1] ]
        i = i + 2
    else:
        i = i + 2   # Skip this hatch segment, but note that we must skip the segment entirely,
                      # not just skip its beginning point

oskay, 2016-01-08:
Thank you @ShelMi, this looks like a definite improvement.

commit 6c01a597 (2016-01-08): Hatch: Correct bug with nonzero gap length, add live preview.

Close #32
closed 2016-01-08 by oskay