evil-mad/EggBot #33: Path reorder fails

krummrey, 2016-01-01:
Whenever I try to reorder paths I get this error: Do I have to preprocess the file in some way?
`Traceback (most recent call last):
  File "eggbot_reorder.py", line 170, in <module>
    e.affect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex.py", line 268, in affect
    self.effect()
  File "eggbot_reorder.py", line 149, in effect
    item = ( id, self.get_start_end( node, transform ) )
  File "eggbot_reorder.py", line 108, in get_start_end
    sp = simplepath.parsePath( d )
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/simplepath.py", line 90, in parsePath
    token, isCommand = lexer.next()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/simplepath.py", line 31, in lexPath
    length = len(d)
TypeError: object of type 'NoneType' has no len()`

oskay, 2016-01-01:
I have not seen this error, and I cannot seem to reproduce it, trying today on a new file.

This is likely due to an improperly formed SVG file, at least according to some definition of "improperly", according to Inkscape. 

I would suggest trying the following, as a first pass: 
Edit > Select all in all layers
Object > Ungroup (Possibly more than once, if you have nested groups in your file)
Path > Object to path

If converting all of your objects to paths does not work, I am very curious how the file was made. Does this have imported artwork? Do you have a procedure to replicate the issue from within Inkscape, starting with a new file?

If all else fails you might try to zip and send us the SVG file (by e-mail) so that we can examine and try to reproduce the issue.

krummrey, 2016-01-04:
The SVGs came all from Illustrator CS6. I had ungrouped them but for some reason I had to do the 
Path > Object to path
to get it to work. Now it works like a charm. 
Thanks for the suggestions.

ShelMi, 2016-08-12:
My experience is identical with krummrey's - both the error and positive results from oskay's suggested fix.  Until I chanced to find this issue and solution in github, I had given up on using eggbot_reorder.  I daresay others may have had this problem and given up also.

Since the fix is so simple and effective, I propose that the eggbot_reorder.py should print oskay's instructions for attempting the fix.  Demo svg file below, after proposed change:

Original:
```
	def get_start_end( self, node, transform ):
		"""Given a node, return the start and end points"""
		d = node.get( 'd' )
		sp = simplepath.parsePath( d )
```
Suggested change:
```
	def get_start_end( self, node, transform ):
		"""Given a node, return the start and end points"""
		d = node.get( 'd' )
		try:
			sp = simplepath.parsePath( d )
		except TypeError:
			inkex.errormsg	(
	'Internal error occurred, please try the following steps:\n\
	1.  Select all objects you want to reorder\n\
	2.  Ungroup [Object/Ungroup] (repeat until "Group" no longer appears in status line)\n\
	3.  Convert to path [Path/Object to Path]\n'
							)
```

Here is a simple example file which demonstrates the problem.
[eggbot_reorder_fail_demo.zip](https://github.com/evil-mad/EggBot/files/414824/eggbot_reorder_fail_demo.zip)

sterlingcrispin, 2016-10-09:
thanks! awesome to run into this fix! This should get added into the Axidraw plugin as well super helpful