evil-mad/EggBot #34: Extensions / Eggbot contributed / Stretch... fails with circle

ShelMi, 2016-01-03:
Stretch generates the following python error statement when I run it on a single circle.  If the circle is embedded in a more complex drawing, stretch still fails.  Circle can be any reasonable size.

I have a proposed fix, listed below the traceback.
Here's the traceback:

Traceback (most recent call last):
  File "eggbot_stretch.py", line 673, in <module>
    e.affect()
  File "C:\Program Files\Inkscape\share\extensions\inkex.py", line 268, in affect
    self.effect()
  File "eggbot_stretch.py", line 654, in effect
    self.recursivelyTraverseSvg( self.document.getroot(), self.docTransform )
  File "eggbot_stretch.py", line 347, in recursivelyTraverseSvg
    self.recursivelyTraverseSvg( node, matNew, v, find_bbox )
  File "eggbot_stretch.py", line 517, in recursivelyTraverseSvg
    self.mapPathVertices( d, node, matNew, find_bbox )
TypeError: mapPathVertices() takes exactly 2 arguments (5 given)

Proposed fix:
Replace line 517 which is currently:

    self.mapPathVertices( d, node, matNew, find_bbox )

With the following:

    self.getPathVertices( d, node, matNew, find_bbox )

ShelMi, 2016-01-04:
Though my proposed fix "works", oskay in his response to issue #35 reported the root cause of this error - I needed to convert the circle object to paths.  Doh!  8^)

oskay, 2016-01-04:
*If* there is a straightforward way to convert the circles to paths before stretching -- and I think that there is -- it is likely worth implementing it.

oskay, 2018-04-04:
Fixed in recent commit.

closed 2018-04-04 by oskay