evil-mad/EggBot #35: Extensions / Eggbot contributed / Stretch... does not stretch spiro paths, stars, or spirals

ShelMi, 2016-01-04:
The stretch extension fails to transform spiro paths (neither bezier nor freehand), stars, or spirals.  No error message is generated.

oskay, 2016-01-04:
You may need to convert the objects to paths first.

ShelMi, 2016-01-04:
Doh!!  You are _so_ right.  My apologies, think I'll adopt radio silence for a while.  8^)

oskay, 2016-01-04:
No silence necessary; I appreciate you mentioning these things, since they are important user-interface issues, even in those cases where the program is likely functioning as intended.

It is likely _not_ easy to convert these types of objects into paths automatically. (For whatever reason, Inkscape extensions cannot call Inkscape's internal functions or menu items), but we should probably do a better job of making this clear to the user.