evil-mad/EggBot #36: Reduce number of pen lifts in hatch fill

ShelMi, 2016-01-14:
A drawback of the current hatch fill is that it takes a very long time to complete if there are a lot of hatch lines.  If the pen lifts could be avoided, the hatch fill would be substantially more usable.

I have been working on this problem, and believe I'm getting very close to a real improvement in plotting speed.  The only problem is, my coding style is not up-to-snuff.  So...question, is there a coding standards document I can refer to?

oskay, 2016-01-14:
Yes, it is quite inefficient in the pen-lift sense. I have thought of addressing that a few times, and would welcome any improvement.

We do not have a particular coding standard that we require (I suspect that my own coding is not up to snuff, by many standards!) but a good starter guide is here: 
http://docs.python-guide.org/en/latest/writing/style/

_Aside_: At a slightly higher level, one might consider using a better filling method altogether. For example, the smooth curves produced by the Hatches (rough) live path effect (detailed at http://wiki.evilmadscientist.com/Creating_filled_regions#Hatch_Method ) actually print much better and faster, because the pen does not have to come to a full stop before turning around. (Another example of this is the one being implemented in the RoboPaint 2.0.0 beta-- you can see a preview in the settings window here: https://github.com/evil-mad/robopaint/issues/232#issuecomment-144303295)
Similar fast-path filling algorithms have been implemented elsewhere (including in python) for use with extrusion-type 3D printers, which essentially do this same thing with molten plastic, rather than ink. I wonder if it might not be straightforward to implement one of those?

ShelMi, 2016-01-14:
Thank you for the pointer to the style document.  I shall read it carefully.

Regarding your aside, I suspect the existing fill method actually is perfectly suited for the task.  And it's a very nice piece of work.  I have (almost) finished my coding for the sharp-corners method, and in my two test cases I have found plotting speed improvements on the order of 50% - 80%.  With luck, tonight I shall be rounding-off the sharp corners, so we'll see what you think then.

If you have any test files you think I should run, please let me know - thanks!

oskay, 2016-01-14:
Yes, rounding off the sharp corners could make a very nice improvement. :)

ShelMi, 2016-01-14:
Here is an early version, warts and all, for comment.  I would appreciate a comment on whether the amount of smoothing I've put on is enough to be kind to the eggbot.  Other comments welcome of course.

[eggbot_hatch_NEW.zip](https://github.com/evil-mad/EggBot/files/90990/eggbot_hatch_NEW.zip)
[Hatch join hasten test.zip](https://github.com/evil-mad/EggBot/files/90992/Hatch.join.hasten.test.zip)

oskay, 2016-01-15:
This looks like a potentially very nice improvement. I really like the curved ends! These should print very efficiently! Let me make a few suggestions, even before looking at the code.

General comments: 

(1.) Rather than posting a ZIP here, it is generally considered best practice to fork the repository (I see that you've already done that), open a new branch for your changes, and then (when ready) open a pull request. You can read more about that process here: https://guides.github.com/activities/contributing-to-open-source/#contributing

For discussing the merits of the proposed changes in general, an issue like this is a great method. The comment thread on a pull request is often a great way to discuss minor issues having to do with formatting and integration.

(2.) As this is a natural evolution of the Hatch fill extension, there is no need to create a different extension with a new name. Please go ahead and keep the same name, so that it replaces the older version in the menu. You can give it a version number if that makes you more comfortable.  (If you would like to keep it separate during development so that it can be compared to the old version, that's totally fine, but we should converge the names when we pull it in.)

[image]

The screen shot above shows what the output looks like, compared to the previous version, with the same general settings. (The "reduce pen lifts") option is selected. Some specific comments about the behavior of the extension:

(1.) The hatches should go in front of, not behind, the existing objects. This is a matter of order in the file. 

(Getting ahead of our selves a bit, the order of paths in the file should be considered very carefully, as a carefully chosen path order can dramatically reduce the amount of "air time" that the pen spends up, traveling between paths that it will draw. The "reorder paths" extension is one way to do this, but we might consider adding that routine for the hatches that are generated here.)

(2.) If you try to ungroup the hatch paths from the object that they are attached to, you will find that they are quite heavily grouped. Taking the lower left object (hollow star) for example, it took 38 ungroup operations to fully separate the hatches. 

What I would hope to see instead is the behavior of the earlier extension: One ungroup operation is necessary to detach the set of hatches from the object, and that the resulting object &mdash; separable from the original object &mdash; is a *single path object*, not a group of paths, nor a set of unconnected objects.

(3.) The set of hatches consists of only 38 hatch paths, which means that drawing it takes 76 pen down or up operations  This is a big improvement over the 75 hatch pieces in the original version, giving 150 pen up or down movements.

(4.) The rounded edges appear to "color outside the lines" -- which actually doesn't look very good when plotting, coloring in shapes. Can the rounding be added *inside* the lines instead?

[image]

(5.) It looks a little haphazard as to which line ends get connected/rounded. It looks as though some lines that have not been could yet be joined, saving many more pen lifts.

ShelMi, 2016-01-20:
Thanks for your encouraging words and enlightening comments to this newbie.

I've been having fun working on the program, and have made real progress.  But, I haven't gotten any traction with github.  I'm totally not "getting" it yet, despite trying tutorials.

Anyway, I want to whet your appetite for my changes by showing the current state:
[image]
[image]

oskay, 2016-01-20:
Beautiful! That looks like a huge step up!

And, absolutely no worries on "getting it" -- we're very happy to help, in any way that we can, and we can indeed accept contributions via zip files. If you're interested in going the more traditional "git" route, we would recommend downloading the github desktop app. It's quite possible to use and work with github, even without knowing any of the command line options.  (Typically, "clone" a repository to download it with the app, make your changes, "commit" them to create a "save point" and then "sync" to upload.)

ShelMi, 2016-01-20:
Yes, I really want to use git.  The only version control software with which I'm familiar was Microsoft's Source Safe.  The advantages of source control, compared with just uploading .zip files, are overwhelming.

That said, I'd be happy to upload the .zips as they are now if you want them.  In fact, now that you mention it, I'll do it right now.  I do realize there are many issues left to be addressed - you'll find the issues I know of listed near the top of the .py file.

[eggbot_hatch.zip](https://github.com/evil-mad/EggBot/files/98116/eggbot_hatch.zip)

ShelMi, 2016-02-03:
Taking you up on your offer of helping me around my not "getting" the git process, I attach for your comments my latest edits to the hatch extension.  So far, the only remaining problem of note is the one referenced in my comments as issue 19.  (I think/hope)
19. Sometimes makes joins that run across a narrow portion that should not be filled.
  		e.g. lower left point of lower left star in EMS test, with 45 degrees, 6.0 spacing, 0 gap, yes crosshatch
  		Status: Open.  Fixing this will increase processing time, as each proposed join will have
  				to be checked against all segments.  May or may not be worth
  				either the programming time or the processing time,
  				as can also be remediated by user choice of slightly different spacing.

[eggbot_hatch 52.zip](https://github.com/evil-mad/EggBot/files/115000/eggbot_hatch.52.zip)

oskay, 2016-02-04:
This seems to mostly work, up to a few little quirks-- its' obviously a huge improvement over what we have had previously.

What is the issue 19 that you're referring to?  
It's not #19 here, possibly this one: https://github.com/evil-mad/EggBot/issues/32  ?

oskay, 2016-02-05:
One quirk-- also present in the original, I believe: It does not run if the page size is defined in inches, rather than px.

ShelMi, 2016-02-05:
When I referred to issue 19, I guess I was off in my own little world.  The number 19 referred to my own list of issues, contained in the comment section near the top of eggbot_hatch.py.

The issue is that when the scope of the segment-joining algorithm is set too high, sometimes a curve can be drawn right across an area that should not be filled in.  It is, I think, fixable, but the fix requires a substantial amount of time/work and before I make that investment I would like to see if the problem affects people to a great degree.

Are there additional quirks which you feel merit attention?

I did not know about the page-size-defined-in-inches quirk.  My work has piggybacked entirely off the original - and I am almost certainly <em>not</em> competent to fix any issues in the original.  I do confirm your observation that the original has a problem when the page size is inches.

ShelMi, 2016-02-05:
I'm inclined to not even offer the option of doing the connection with straight lines, and just assume the user wants a nice curvy output - what do you think?
[image]

oskay, 2016-02-05:
On the units: Let's worry about that later. I've been fighting similar issues in other contexts, and should be able to help.

There are some additional quirks-- sometimes not filling certain objects, for example. I haven't found a reliable example to give you yet. I also need to look at the grouping structure a little more. I will get back to you on that.

I'd say go ahead and remove the option; this is a much better default.

Unrelated thing that might be worth considering, if you find that it might be straightforward: 
When filling in a shape, we often copy and inset that shape first, to make the filling lines *not quite* go all the way to the edges. When using pens of finite width, that often helps to make a much cleaner shape, since it always "colors within the lines."  Since you are already offsetting the straight line segments from the edges in order to allow room for the curves, it might be possible to add an option to inset the fill a little bit from the edges that you are filling.

ShelMi, 2016-02-05:
When you find examples of not filling certain objects, it's worthwhile to compare to the identical fill using just the original method.  As I say, I totally rely on the original method to do the heavy lifting.

I'll remove the option.  Should I leave in the SAVE YOUR FILE FIRST message, do you think?

I like the idea of the insetting option.  I don't know what worms might crawl out of that can - I think I'll just have to go ahead and try it, and we'll see what problems it does or doesn't make.

ShelMi, 2016-02-05:
When I just now tried my test image with inset edges, I ended up with nasty problems in many places.  I need to get these problems squared away.  It definitely does not look ready for prime time yet, despite my earlier hopes.  I'll be working on it.

ShelMi, 2016-02-05:
I'm being over-hasty.  When I made my comment above, I thought the badness I was seeing was a reflection on the viability of my current version.  Turns out it was not a problem with the current version, it was a problem with my attempt at a super-quick implementation of insetting.

ShelMi, 2016-02-05:
Hey, wait a minute??  As I've been pondering the insetting problem, I've come upon a question...Have users been advised to do an inset operation before hatch filling?  If so, I don't see how that can work properly when there is a shape with another shape inside.  I've tried to illustrate what I mean by the attached image, which is of a green original square with interior green square.  Selected all.  Duplicated. Inset. Stroked red.  Original-hatch filled.

In other words, nested objects need to alternately by inset/outset to make this be a good way to constrain the hatch domain of action.

I'm not saying I can't do things right, but I'm sure there will be edge cases where it doesn't behave as desired.

What do you think?  Have I just got fog in my cockpit?

[image]

oskay, 2016-02-05:
> When you find examples of not filling certain objects, it's worthwhile to compare to the identical fill using just the original method.

Yes, of course.

>Should I leave in the SAVE YOUR FILE FIRST message, do you think?

No.

I took one of your filled regions, and clipped out the curved ends.  Here is the result:

[image]

It is neatly inset where the curves were-- just not towards the ends, where the curving breaks down. 

For reference, here is what I would consider an "optimal" fill of that region:

[image]

oskay, 2016-02-05:
> In other words, nested objects need to alternately by inset/outset to make this be a good way to constrain the hatch domain of action.

I don't think that this is the case. If you have a single path object (the kind that can be filled), then the Inset operation should enlarge one boundary and shrink the other. 

Example, green original, black inset:

[image]

The advantage of insetting (in practice) is that when you draw the resulting shape with a pen, the fill is less likely to bump outside the outline:

[image]

ShelMi, 2016-02-10:
[eggbot_hatch rev 65.zip](https://github.com/evil-mad/EggBot/files/124371/eggbot_hatch.rev.65.zip)
You can see that I have added a few options since we last communicated:

1.  Hold back hatch from edges is the feature you suggested.  It has several situations where it intrudes further into the border than it should, and at least one situation where it leaves too much room.  Details if you want them.  On balance, I'm pretty pleased with how my simple-minded algorithm worked.

2.  Hold back distance is added because, in conjunction with the new "width of displayed stroke" the user can semi-realistically view and play with the plotted results.

3.  Allowing simulation of the resulting print, by adjusting displayed stroke width, may give useful feedback.  At least, I liked it.
[image]

I was able to come pretty close to your optimal fill of a circle.  Had to play with the "scope of reduced pen lifts" parameters a bit.  Haven't tried it on any random circles, I expect it won't always be so optimal.
[image]

oskay, 2016-02-11:
That's looking really nice-- I haven't had a chance to try it yet, but I will do so shortly, and hopefully have a fix so that it works across multiple units as well.

ShelMi, 2016-02-12:
What is the issue with multiple units?

oskay, 2016-02-12:
(Page units of inches and so forth.)

ShelMi, 2016-02-12:
Oh, good.  In the meantime I'm working on what I hope is the last important problem remaining, illustrated by the deliberately pathological case below:

[image]

oskay, 2016-02-12:
I'm attaching a revised version, with some changes-- mostly cosmetic. I've made changes to the description and option names in the GUI.

I would like to suggest that the line width option be removed, and to instead use whatever the parent object's line width is.

I've added an updated "getLength" command that handles other types of units. I have not yet found the issue that is preventing it from working with pages in inch units. Will keep looking.

The "inset" (formerly, "hold back") works reasonably well. It would be very nice if it could hold back from edges that it encounters in the perpendicular direction as well-- see attached image. However, this sounds harder to accomplish. ;)

[image]

[hatch-revised.zip](https://github.com/evil-mad/EggBot/files/128866/hatch-revised.zip)

oskay, 2016-02-13:
It seems to be working on inch-sized pages (as of that last version), although inconsistently. Trying to nail this down a little further.

ShelMi, 2016-02-13:
I look forward to merging your changes.  Do you have a preferred merge facility, or shall I do it manually?

Wow - that looks terrible!  I have been working on something which will fix the gaps in the hatch, but the current presentation looks truly raggedy.

As far as insetting from perpendiculars, I believe it _may_ be doable.  I had been hoping to avoid it, thinking that the user will likely only be insetting enough to avoid any little hooplas ouside his border - sort of a half-pen-width inset.  And thus a single hatch too close, but parallel to, a boundary would likely be acceptable.

I'd like to continue working on the fix which accounts for the gaps, before switching my attention to the perpendicular inset problem.  That's assuming you think from your knowledge of how people use this that it (perpendicular inset) needs to be fixed?

Speaking of things working inconsistently, I'm rapidly coming to the conclusion that inkex.errormsg has a limited buffer space, and that writing too much debug stuff out causes Inkscape to take a journey into hyperspace.  Do you find this to be true, or have I misled myself?

Good luck on the size issue.  When does it happen that people use inches as their units?

oskay, 2016-02-13:
Without any doubt, the basic "perpendicular" offset that you've implemented thus far is a *huge* improvement.  Detecting the other edge would be even better, of course-- but yes, it's absolutely a huge improvement thus far.

I have not had any issues with large inkex.errormsg outputs, _except_ that they are hard to read.  I'm attaching an example of a debugging extension (not guaranteed to run!) that writes a log file to your home directory. If you look at how this one works, you'll see that this method is much more suitable for large quantities of debugging data.

Found an odd issue when filling on a page with inch size units. Screenshots attached.  This is on a page with the Letter (8.5 x 11 inch) template, and a simple 5-inch rectangle. Works differently with the option checked, as you will see.

[image]
[image]

[Archive.zip](https://github.com/evil-mad/EggBot/files/129324/Archive.zip)

ShelMi, 2016-02-14:
Huh, I'm confused - the first hatch fill of the two above looks like the "hatch (rough)" method - the hatch lines are not straight, and the joins at the end are pointier.  How can this be???

oskay, 2016-02-14:
I'd advise you to try it yourself, and see if you can replicate it under the conditions that I described.

ShelMi, 2016-02-14:
I still don't "get" it.  8^)
[image]

oskay, 2016-02-14:
Interesting! I wonder if we're off by a few details in our respective codes.

ShelMi, 2016-02-14:
Attached files are where I currently am - I think _maybe_ I've finally licked the "partial blank hatches" problem.

Regarding the "odd issue", I cannot think of any way for my algorithm to generate output that looks like the "hatch (rough)" method.  I sure hope it's just some kind of mixup.  Can you still replicate the results?
[eggbot_hatch 66.zip](https://github.com/evil-mad/EggBot/files/129514/eggbot_hatch.66.zip)

oskay, 2016-02-14:
Same behavior with v 66-- I'll follow up later.

oskay, 2016-02-14:
Of note: If I start with a generic new document, and then change the size (using document size) to 8.5 x 11 inches (or click letter there), it does not cause the issue.  It is only when starting from a fresh new document from Templates > Letter.

ShelMi, 2016-02-15:
OK, using the "starting afresh" method, it now reproduces here.

ShelMi, 2016-02-15:
Wow - the thing that looks just like the "hatch (rough)" turns out to be the bezier control points being displace a long way beyond what the should be displaced.  Is there any way that the work you've been doing on dimensions somehow does something different to bezier control points, which are in the path as an element of designator type "c"?

ShelMi, 2016-02-15:
OK, thanks to your "of note", herewith iteration n+1

I have _not_ implemented the "other direction" inset, and it may be quite a while before I feel I could do a good job on it.

I hope the attached survives the evil mad scientist torture better than the previous version did!  8^)

[eggbot_hatch 67.zip](https://github.com/evil-mad/EggBot/files/131617/eggbot_hatch.67.zip)

ShelMi, 2016-02-17:
I have become aware of a second, probably less important, issue which can show up when insetting the hatch.  Best illustrated by the two images below.

This first image shows the issue behaving badly:
[image]

And this second image shows the issue not happening, because min gap is set low enough to avoid the part of the logic which caused the problem shown in the first image:
[image]

And here, I think, is the fix for this issue:
[eggbot_hatch 68.zip](https://github.com/evil-mad/EggBot/files/133658/eggbot_hatch.68.zip)

oskay, 2016-02-17:
You have nailed that-- the small gap option was a stopgap measure added to reduce the number of pen lifts. Under the circumstances, I would advocate removing it completely.  Looks like it might be a good fix for that issue.

ShelMi, 2016-02-17:
A "stopgap" measure indeed - perfect definition!!

So...does that mean I am mandated to stripping the gap logic completely out?  (As noted above, I think I successfully exorcised the latest gap-related issue.)

oskay, 2016-02-17:
Yes-- pun not intended-- eek!

Yes, please strip out that logic entirely. :)

ShelMi, 2016-02-18:
Bye bye minimum gap.
[eggbot_hatch 69.zip](https://github.com/evil-mad/EggBot/files/135754/eggbot_hatch.69.zip)

ShelMi, 2016-02-22:
Found a bug that I think probably predates my messing around with hatch fill (though I haven't actually checked it):
   If you attempt a hatch fill on either an empty document, or on a selection which is not closed, then Inkscape spends forever spinning its wheels while doing and saying nothing.  User loses any unsaved work and must forcibly end Inkscape.
   Proposed fix attached.
[eggbot_hatch 70.zip](https://github.com/evil-mad/EggBot/files/140271/eggbot_hatch.70.zip)

oskay, 2016-02-22:
That sounds like a serious bug indeed-- and it wouldn't surprise me entirely, either. It's just the kind of thing that might not actually get tested.

When trying the crosshatch option, on some basic shapes, I was able to get the following error:

```
Traceback (most recent call last):
  File "eggbot_hatch.py", line 1731, in <module>
    e.affect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex.py", line 268, in affect
    self.effect()
  File "eggbot_hatch.py", line 1178, in effect
    float( self.options.hatchSpacing ), False )
  File "eggbot_hatch.py", line 1098, in makeHatchGrid
    retValue = bBoundingBoxExists
UnboundLocalError: local variable 'bBoundingBoxExists' referenced before assignment
```

ShelMi, 2016-02-22:
Thanks for finding that so quickly!

This may fix it:

[eggbot_hatch 71.zip](https://github.com/evil-mad/EggBot/files/140492/eggbot_hatch.71.zip)

ShelMi, 2016-02-23:
I'll start by acknowledging that the way I use Inkscape to design a print may be atypical.  When I do a design using hatch fills I repeatedly find myself doing first the hatch fill, then selecting the black hatch and changing its color to approximate the color of the pen I'm planning on using for that fill.  (I use a swatch palette that embodies the colors of the pens I have.)

So, from my perspective I think I would like to have the option of specifying that the displayed hatch color should be the same as the boundary color that is being filled, rather than always being black.

Does this make sense to you?  I've already experimentally made such a change for my own use, but have not done any designing since to see how I like it.  Any thoughts?

oskay, 2016-02-24:
This is all looking quite nice. Still some slightly imperfect behavior when filling unusual shapes, but it's really quite an upgrade in usability.

[image]

On the fill color/width/etc: I understand where you are coming from. My instinct is that we should pick exactly one behavior, rather than give any UI options. Inkscape has a huge range of line styles, and we can only add an interface to very few of them in the space given. 

Here are two approaches that might be worth consideration:

* I had suggested inheriting the stroke options from each parent object-- we're already filling each object, so have access to their SVG properties. This works well for most cases, and emulates the behavior of the Hatches (rough) path effect. It would also ensure that the color follows in a natural fashion. 

* Another approach would be that we could allow one to start with filled objects. If an object were to have a solid-red fill, for example, our hatch lines could be added with the same thickness as the object's stroke (if nonzero), and in the same red color as the fill. We could then remove the solid fill.  To the end user, this could be quite intuitive-- a solid red object becomes an object with a red (path-based) fill and stroke.

ShelMi, 2016-02-25:
> Still some slightly imperfect behavior when filling unusual shapes, but it's really quite an upgrade in usability.

Your example looks like an issue that I thought I had fixed.  I'll have to get back on it.

> On the fill color/width/etc: I understand where you are coming from. My instinct is that we should pick exactly one behavior, rather than give any UI options. Inkscape has a huge range of line styles, and we can only add an interface to very few of them in the space given.

And I strongly believe in following instinct, especially the instinct of a person with such a broad understanding of the customer base.

My preference of your two suggested approaches is the first one.  Regarding the second, I would guess that turning a fill into a hatch would encourage users to be a little sloppy - because it might be hardly noticeable that anything had happened and they might therefore get an unwelcome printing surprise when they forget to do a hatch fill.

So...what I'd like to do is:

1. Try to smack down the bug that gave the bad result in your example.  I know there will be plenty of other imperfections, but I'd like to at least get out any where I think there is a decent solution.
2. Try to implement your suggestion of inheriting stroke options from the parent object.  This will be an uphill struggle for me, as I do not yet rise even to the "duffer" level in my understanding of python or of SVG.  I welcome the challenge however, it's just that my coding is likely to be very awkward.  If that's OK with you, it's OK with me.

oskay, 2016-02-29:
Yes, that sounds like a great approach. 

If you run into trouble with setting the stroke options, I might be able to lend a hand. To look at how one might copy and paste the style from one object to another, you might look at the Interpolate extension, found under Extensions > Generate from Path. This extension can create interpolated paths that are the weighted average of two initial paths, with different styles.  You can browse its code, too: interp.py

ShelMi, 2016-03-01:
Super, glad you like the way it's going.  Thanks much for the pointer to code relevant to style, it'll be a great way to get started.

Attached is a version which attempts to fix the "slightly imperfect" behavior you demonstrated earlier.  It only addresses that behavior, and not any of the other problems such as hatches close-to but not intersecting segments.

This version is just for any testing you may choose to do regarding the imperfect behavior, and does _not_ have any attempt to inherit stroke options.
[eggbot_hatch 74.zip](https://github.com/evil-mad/EggBot/files/152211/eggbot_hatch.74.zip)

ShelMi, 2016-03-01:
Oh, color and width weren't as bad as I feared.  I had already stumbled (and I do mean stumbled) through the color, and width was easy after that.  I do not have a clue, however, whether I have implemented it in an appropriately conservative way.  If you choose to take a look, it's down in the vicinity of line # 1100.

[eggbot_hatch 75.zip](https://github.com/evil-mad/EggBot/files/152326/eggbot_hatch.75.zip)

oskay, 2016-03-03:
On my first try, I get the following error:
```
Traceback (most recent call last):
  File "eggbot_hatch.py", line 1777, in <module>
    e.affect()
  File "/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/inkex.py", line 268, in affect
    self.effect()
  File "eggbot_hatch.py", line 1526, in effect
    self.joinFillsWithNode( key, stroke_width, path[:-1] )
  File "eggbot_hatch.py", line 1102, in joinFillsWithNode
    stroke_width = float(val.strip())
ValueError: invalid literal for float(): 1.39557528px
```
I would suggest looking more closely at how the interpolate extension manages it-- using the "has_key" and "deepcopy" methods, for example.

ShelMi, 2016-03-04:
I really appreciate your patience and suggestions.  If I don't take your suggestions it's because I can't figure out how to, not because I don't want to.  My background is so strongly in assembly language, C, and Visual Basic that I just can't seem to wrap my head around .SVG, .XML, Python, etc.  ("Yet", I add hopefully.)

That said, I hope you can humor me just a bit longer:
1.  Can you explain how I can re-create the error you got?.  I've tried, in Inkscape, setting the stroke dimension to inches, mm, px, % and can't re-create the issue.  I've also set document properties to inches and default units to inches, again with no luck.
2.  Could you try the attached version 76 to see if by some stroke of divine providence it functions correctly?
[eggbot_hatch 76.zip](https://github.com/evil-mad/EggBot/files/158189/eggbot_hatch.76.zip)

oskay, 2016-03-04:
Divine providence has struck. :)

In the settings, `holdBackSteps` could be made to have a lower range. I suggest `min="0.1"`.

It looks like the holdback may be a little too aggressive on dropping lines.

[image]

This example has a holdback of (I think) 1 px-- just to make the inset lines not exceed the edges when using a fat pen. However, it seems to also be missing a lot of fill lines that should have been able to fit in there.

oskay, 2016-03-04:
It looks like things may not be properly adjusting to pages with different units-- If I take some sample objects, and paste them on to new documents using different templates ("plain" new document, "Letter", "A4", "EggBot", etc), the results from filling them with hatches should be the same everywhere. However, it appears that it sometimes works strangely (A4), and not at all on others.

ShelMi, 2016-03-05:
1. OK on the holdback, will do, but also see 2.

2.  Addressing the "aggressive holdback" issue, it goes more deeply than the holdback value.  There was a plain old bug whereby the program was demanding there be space to insert a bezier curve - even when not connecting nearby ends.

3. When fixing the issue in 2, a new question arises:  What is the shortest hatch which should be drawn?  I tentatively propose it should be related in some way to the hatch spacing.  Presumably user has chosen a spacing that bears some relation to the pen tip width and egg size, so the spacing and/or holdback may be usable as a metric.

4. Your example prompted me to think some more about my procedure when connecting nearby hatches.  I could, presumably, fit in more joined edges if I made the Bezier curve less gentle (only where required).  Do you think that worthwhile?  And see 5.

5. The question in 4 arose because right now, when joining hatches, I completely discard any hatches that could be joined except that they do not have room for the Bezier curves.  Do you think I should, instead, plot said hatches as straight lines, rather than discarding them?

6.  There's probably more, but the programming part of my brain is oxygen depleted now, so I need to deal with 1 - 5 before I get too involved with the (ugh) units issue.

oskay, 2016-03-05:
> What is the shortest hatch which should be drawn? I tentatively propose it should be related in some way to the hatch spacing. 

That's a fine point. Looking at the "Text" example above, I would say that either something like (hatch spacing)/4, or perhaps equal to the line width would be appropriate choices.  Another approach would be to make it a user-controlled variable, of course.

> I could, presumably, fit in more joined edges if I made the Bezier curve less gentle (only where required). Do you think that worthwhile?

Yes, I think that it is. The big time savings is coming from the lack of pen up-down is much more important than the less-than gentle edge. I should also note that we're working on some core-engine improvements that will lessen the effect of less-than-gentle curves. :)

> Do you think I should, instead, plot said hatches as straight lines, rather than discarding them?

Interesting! I've seen curves that are arbitrarily flat in places, so it's hard for me to picture this circumstance exactly... If you had to choose between connected and _exactly_ straight, I'd say that it's close, but probably better to connect them.

> so I need to deal with 1 - 5 before I get too involved with the (ugh) units issue.

I can also probably help with this-- as some of our other routines work with different units. Last time we ran into a version control issue-- not too easy to suggest changes when both versions are changing. :)

ShelMi, 2016-03-05:
> I would say that either something like (hatch spacing)/4

Done

> Yes, I think that it is. The big time savings is coming from the lack of pen up-down is much more important than the less-than gentle edge.

Thanks for that input - it has resulted in what I think is a much more thorough hatching.  I now make the joins get progressively less and less curvy as the available width shrinks.  In NH anyway, it works really well.  We'll see about CA.

I think/hope this disposes of 1-5.

> I can also probably help with this-- as some of our other routines work with different units. Last time we ran into a version control issue-- not too easy to suggest changes when both versions are changing. :)

That would be great, but I worry about burdening you with extra "stuff" to do on top of your evil mad scientist activities.  Do you want to give me a couple of days to see if I can do anything with it?  I'll probably start poking at it anyway - at least I may learn something.  8^)

If you do decide to make changes, and if I have made changes to hatch algorithms at this end, I suspect the changes will be fairly orthogonal to each other and thus that I could do a merge using windiff.

Attached is version 80.

[eggbot_hatch 80.zip](https://github.com/evil-mad/EggBot/files/160060/eggbot_hatch.80.zip)

[27 more replies]

closed 2016-03-18 by oskay