evil-mad/EggBot #37: eggbot_stretch works well for spheres, but...

ShelMi, 2016-03-02:
This is not, by a long shot, a big deal - but I think the stretch algorithm could have just a little bit of refinement added to it.  The current version, nice as it is, nonetheless maps the stretch function as though the print were on a sphere.

I wonder if the data/methods presented at http://www.takayaiwamoto.com/Egg_Draw/Real_Chicken_Egg.html might be used to do a more accurate mapping of SVG coordinates to the egg ovoid.  A rough test, based on the below image, shows that the differential error between a sphere and the large end of an egg may be as much as 33%.

I say this in the full realization that +/- nobody will really notice the difference, but I still would be happy to take a shot at it.  If further data presents itself, one might even consider having separate settings for ostrich, duck, chicken.

[image]

oskay, 2016-03-02:
It does indeed model a sphere. However, often times, that's exactly what people are printing on.  Also, we have learned over the past couple of years that there is tremendous variation, even in just chicken eggs. :)

If we add support for proper egg shapes, then...
* It will add some complication to the UI, if we add additional options. It might be better relegated to a _separate_ extension ("Advanced egg stretch"?)
* Most people won't notice the difference. This is a "second order" correction, after the horizontal stretch for the aspect ratio (motor axis offset) and for the main stretching.
* It will be challenging to tell if the stretch is "right side up" -- with the symmetric stretch that we have right now, it works reasonably well with the egg facing either way in the machine.

None of these is to say that it shouldn't be pursued, just some food for thought.

ShelMi, 2016-03-25:
Points taken.

I have a substitute suggestion which I have been using for a few days, and I like having the additional flexibility that comes with the options in this substitute "advanced stretch".

I still am thinking about a user-friendly method for un-egg-shaping the plot, but that will have to wait for a future day.  8^)

[image]

[eggbot_advanced_stretch 01.zip](https://github.com/evil-mad/EggBot/files/189090/eggbot_advanced_stretch.01.zip)

ShelMi, 2016-03-29:
I needed greater-than-integer percent precision when I was doing a stretch, so have a new version with no other changes than allowing up to 4 digits past the decimal point in the expand percent field.

[eggbot_advanced_stretch 02.zip](https://github.com/evil-mad/EggBot/files/192983/eggbot_advanced_stretch.02.zip)

oskay, 2016-03-29:
I would suggest splitting your GUI into at least two tabs, to make the interface smaller. Perhaps a "description" and "help" tab in addition to the main "stretch" tab.