evil-mad/EggBot #38: EggBot_250A.exe Windows installer fails if "Inkscape Extensions" unchecked

EmbeddedMan, 2016-03-24:
If you run the EggBot Windows installer, and uncheck the "Inkscape Extensions" box in the "Choose Components" window but leave the "USB Driver" box checked, the installer will generate the following errors when you click "Install":

"Error opening file for writing: \Driver\mchpcdc.cat"

I think what's happening is that the installer doesn't know how to unpack and run the USB Driver installer files if the "Inkscape Extensions" don't get installed first. This happens even on machines where Inkscape is already installed. My guess is that there is something in the installer script that the USB Installer section needs, but it is turned off if the Inkscape Extensions are not installed.

Tested on Win7, Win8.1 and Win10, and they all appear to behave the same.

oskay, 2016-03-24:
Thanks for catching it. We're due to make a new installer anyway. 
Was hoping to add that one new feature before rolling new installers....

oskay, 2019-06-21:
fixed in v 2.8.1 installer, releasing this week.

closed 2019-06-21 by oskay