evil-mad/EggBot #39: Suggestion: Provide USBDriverInstaller as separate download

EmbeddedMan, 2016-03-25:
(Windows problem only) If somebody wants to use an EBB without Inkscape, they currently have to download EggBot_250A.exe on a machine that does have Inkscape, run it, then go to C:\Program Files\Inkscape\Driver and copy that folder over to the machine without Inkscape so they can run USBDriverInstaller.exe to install the EBB drivers.

Would it be possible to package just the Driver directory in it's own ZIP file which can then be made available for download separately? It would not need to be part of the EggBot repo so that it doesn't confuse folks who are downloading the whole EggBot software package.

oskay, 2016-03-25:
Not a solution, but it is still available at: 
https://code.google.com/archive/p/eggbotcode/downloads

Part of the problem is that we have trouble communicating which software *should* be downloaded. We might want to add a github.io page to help communicate what to download more clearly.

oskay, 2016-10-26:
Added to release Version 2.7.1.

closed 2016-10-26 by oskay