evil-mad/EggBot #42: Feature request: A pen alignment method might be useful

ShelMi, 2016-04-12:
I assume I'm not alone when it comes to using more than one brand of pen on the same egg.  For example, I like to use a Copic 0.2mm marker for the black, while using Sharpie ultra-fine for the colors.

I envision a system where user gets to specify an unused section of the egg as an alignment test section.  User goes there with each change of pen, and walks pen/egg until a pen "peck" lines up with the peck from previous pen.

As luck would have it, I've been working on just such a feature for myself - if it would be useful, I could submit code.

oskay, 2016-04-12:
We have sometimes done this manually, with a separate "registration mark" layer, and adding some manual walks.  What did you have in mind as a way to do this more automatically?

ShelMi, 2016-04-13:
Oh, it would be so nice if it could be automated - but I can't think of any way.  I'm not sure what your exact registration procedure is, but I imagine it's rather like what I'm doing here - it's just that I have systematized it by putting it all on one tab.  Only one thing I've added is the adjustable pause to allow user time from when he/she clicks mouse on "apply", to when the pen actually pecks.  This gives time for user to get eyeballs onto egg, maybe even with magnifying glass if desired.

[image]

[image]

oskay, 2016-04-13:
Might it be better to add this as a single addition to the pop-up list on the Manual tab?

List item name: "Test Pen Alignment"
It would:
* Move the pen to the alignment point
* Make a test peck 
* Move back

The actual alignment mark (fiducial) might be best as a little cross ("+"), maybe 5 steps wide x 5 steps tall, or possibly as a circle, say 10 steps in diameter.

If no layer named 'Align" were found, it could (gracefully) give you a dialog box:

> To use the Test Pen Alignment feature, make a layer in your document called "Align" that contains exactly 1, 2, 3, or 4 circles, and nothing else. A small "plus" mark ("+") will be drawn at the center of each circle.

ShelMi, 2016-04-15:
Point well made about this logically being on the Manual tab.

I've been rehearsing this scenario in my mind, and wonder if it's a little awkward in a couple of respects:

1.   When the pen moves away, does a peck, and then moves back, it may be moving back to a position where the user may be unable to see the peck mark.  If, on the other hand, the pen stays in peck position then it is guaranteed to be in a highly visible position - at the action end of the pen.

2.  I find the pre-peck pause highly useful - I find it really helpful to be able to see the pecking action, which I can't do if the peck happens immediately.

My thinking on a "peck" was that it would make minimal visual disturbance on the egg, but I assume you have found that a "+" is better as a fiducial mark?

And I wonder if you would fill me in on how one uses more than one fiducial.