evil-mad/EggBot #45: Tiresome extraneous warnings during single-layer plots

ShelMi, 2016-06-08:
In the following situation the user gets multiple extraneous error warnings:

- Using the "Layers" tab
- There is a _non_-numbered layer with a non-plottable item on it, perhaps an image or some text

Each time user plots a numbered layer, a warning dialog is presented to inform user that the non-plottable item is non-plottable.  This occurs for each and every numbered layer user plots, despite the fact that user has not numbered the non-plottable layer and therefore is not requesting it be plotted.  This gets rather "old" rather quickly, at least when user has more than a handful of pen changes.

A proposed fix is attached.

In addition, this fix also attempts to clarify proper warnings of numbered non-plots, by identifying the name of the offending layer.  An example of a possibly-cryptic warning is:  " unable to draw flowRoot object, please convert it to a path first."  At any rate, it is cryptic to those such as I who do not immediately recognize the term "flowRoot".

[eggbot.zip](https://github.com/evil-mad/EggBot/files/303932/eggbot.zip)

oskay, 2016-06-08:
Excellent catch. And yes, clarifying the flowRoot error is a great idea.

oskay, 2016-06-19:
Implemented. I've gone ahead and trapped the flowroot error directly as well.

ShelMi, 2016-07-06:
The current version of eggbot.py in github, # Version 2.7.6, dated June 19, 2016, continues to have the tiresome extraneous warnings described in the initial issue report.

I propose the changes indicated in the image below: (it's so easy for things to get lost in the shuffle  8^)

[image]

oskay, 2016-07-06:
Right. It was a little tricky to merge this; IIRC, it wasn't exactly in sync with the development version when I added the changes.

*However* I wonder if it might be better for us to wrap yet more of this big "if" structure in an "if plotCurrentLayer" type structure. I think that we could actually stop parsing the element right before the `elif node.tag == inkex.addNS( 'path', 'svg' ):` line, if the layer is not to be plotted. Doing so would not only avoid the extraneous warnings, but would also make it faster.

ShelMi, 2016-07-07:
I understand it is better to treat problems at a higher level if possible.  I couldn't understand where you suggested I put the "if" in the long chain of "elif"s, so I put it where I thought it might be best  YMMV.  8^)  (And removed the tests in the "elif" chain.)

I've given this a bit of testing, but not exhaustively.

[image]

oskay, 2016-07-08:
I've made the change that I was suggesting yesterday-- in my limited testing here, it appears to do the trick. Closing again; let's re-open this if I haven't done it correctly. :)

closed 2016-07-08 by oskay