evil-mad/EggBot #46: Communication error could be salvaged in some cases

ShelMi, 2016-06-08:
Currently if there is a persistent communication breakdown in plotLineAndTime which ends up returning a null from ebb_motion.QueryPRGButton(self.serialPort), then python aborts the plot with a runtime error message. This loses position reference for the plot, and thus the entire plot is ruined.

The suggested change incorporated in the attached file explicitly looks for a null response.  Here if a null response occurs, the program substitutes a simulated push of the pause button.  Therefore the user may be able to use the "Resume" to rescue the plot, after fixing the communication problem.

The attached file is built from a base of the proposed fix for issue 45.

[eggbot (2).zip](https://github.com/evil-mad/EggBot/files/303945/eggbot.2.zip)

oskay, 2016-06-20:
I'm not 100% certain that I've actually seen this issue, but your proposed fix is straightforward and plausible.

closed 2016-06-20 by oskay

ShelMi, 2016-10-28:
Now that ebb_serial.py has an errormsg associated with receiving a null response from comPort.readline(), I have found an exceptionally simple and reliable way to cause the error to occur, so hopefully the root cause may be discernible.The present error is _not_ a hardware or communication problem per se, since it can be reliably caused by merely changing certain parameters on the Timing tab of the Eggbot Control extension.

oskay, 2016-10-28:
This is great-- I'll take a look shortly.

oskay, 2016-10-29:
Able to reproduce. 

May need to add a time.sleep command for longer duration pen up/down moves, much as we do for longer stepper moves.

oskay, 2016-10-29:
Sure enough. 

It turns out that while we execute a `time.sleep` command to pause while a long stepper movement command executes, we did not have the same procedure in place for long pen-up and pen-down movements. I have added a sleep command for the pen-up and pen-down movements which appears to correct the communication error in this case.

ShelMi, 2016-10-30:
Speaking of delays, I'd like to raise a question here rather than on the forum, because if I'm thinking wrongly about it I don't want to confuse the broader community of users.  (Also, if your ```time.sleep``` command changes the behavior I don't want to publically raise an issue which may be going away shortly):

In the eggbot wiki, at http://wiki.evilmadscientist.com/Eggbot_Control it is stated:
"After the pen has been lowered, it is important to wait a fraction of a second before starting to draw again. This gives the pen arm a chance to settle in and recover from any minor bouncing. If you start to draw too soon after lowering the pen, there can be some skipping of the pen tip. To control the length of this pause, the "Delay after lowering pen" value is used. This value has units of milliseconds (ms) where 1 second is 1000 milliseconds. This delay should probably be no less than a quarter second (250 ms) and values as large as half a second (500 ms) are not unreasonable. However, you don't want to use too large of a value as <B>the total delay caused by lowering the pen will be this value plus the time it takes to lower the pen</B>." [emphasis added]

To me, this says that the "Delay after lowering pen" starts at the _completion_ of the servo's move to the down position.  But my observation tells me that in fact the delay starts at the _beginning_ of the servo move.

Iff I'm correctly observing, then the wiki advice to have a delay of no less than 250 ms is cutting it rather closely.  E.g. for U-D = 6 % points, and lowering speed of 20 points/second, the delay expires during the 6/20 = 300 msec servo down motion.  True, the pen might conceivably have gotten to the egg already, but there is nonetheless minimal-to-zero delay before the line commences drawing.

I'm not trying to pick nits here, but iff my observation is correct, would you consider re-wording the wiki description - or okaying me to do it since I'm sure there are innumerable more pressing demands on your time?  Or something else?

oskay, 2016-10-30:
Yes... I *think* that both the behavior is correct, but the description should probably be updated.  The delay does indeed start when the pen begins lowering.  Please go ahead and edit the description to reflect reality. :)

Longer term, I would like to update the pen lift and lowering to use the newer method implemented for the AxiDraw software, where the transit time is calculated automatically (from the speed and distance) and the delay time is added to that. The software change is relatively straightforward, but it also requires changing the documentation, which is a little more time consuming.

ShelMi, 2016-10-31:
Just a note to say I've made the change in the wiki, in case you want to review it for clarity or accuracy.

oskay, 2016-10-31:
Thanks! I've made some minor tweaks for tone, but this is genuinely much better. :)

ShelMi, 2016-10-31:
Glad you like it!  8^)  I see you tweaked only section 7 and not section 9 - perhaps section 9 is feeling left out?

oskay, 2016-10-31:
Ah-- missed that. All tidied up now. _Thank you again._

commit 9355395f (2016-06-20): Handle communication errors when checking button status

Closes #46