evil-mad/EggBot #49: Very very minor issue: X-wrap will be ignored sometimes

ShelMi, 2016-07-06:
I mention this issue not because it's important, but because it involves pen up/down logic and I just want to check that there is no larger problem:

- When the pen is plotting at a distance from the document border - for example when doing a spiral-text plot - and
- if user presses the "pause" EBB button, and
- user then selects "Resume" tab, and
- user checks "Cancel and return home only"
then
- the self.bPenIsUp variable is uninitialized, and therefore
- plotLineAndTime() will ignore user's x-wrap selection, and
- will perform as many egg revolutions, at pen _down_ speed, as it takes to get back home.

A possible fix is shown in the attached image, but it may be that this is but a bandaid for a larger problem?  or a different problem??  Dunno, but just wanted to bring it up.

[image]

oskay, 2016-07-06:
I think that I remember seeing this once in practice, years ago, but had forgotten about it. Thanks for noticing; I'll take a look. :)

oskay, 2016-07-08:
I've made the change that you suggest above. I think that this will fix the problems both with the speed and wrapping, since the wrap-around logic should automatically implement once we're in pen-up mode. Please give it a try.

ShelMi, 2016-07-08:
OK, have tried it on my test file - works like a charm.  Not sure that it needs further testing, what do you think?

oskay, 2016-07-10:
I think that's a wrap. :)

closed 2016-07-10 by oskay