evil-mad/EggBot #51: Set Engraver (SE) command is not synchronized with the stepper motors

zemosh, 2016-08-04:
Regarding timing, the Set Engraver (SE) command behaves differently from the Set Pen (SP) command and the Stepper Motor (SM) command because it is executed immediately. Therefore the engraver is not synchronized with the movement of the stepper motors when more commands are sent to the EBB at once. It would be nice to set PWM for the engraver using the CommandFIFO as it is in the case of SE and SP because then it would be possible to change PWM for the engraver along the path.

oskay, 2016-08-04:
The intent of the engraver option, as currently implemented, is to turn on the engraver before starting to plot and to turn it off when finished. For the application that you are describing, you will probably need to use the QM command to wait for a given motion to complete before adjusting the value.

It might be worth exploring the idea of adding an additional, queued command that can give an IO pin/PWM change... perhaps as an optional argument to the SE command or as a new command altogether. @EmbeddedMan?

EmbeddedMan, 2016-08-04:
I like that idea a lot - an optional parameter to SE that will put it in
the motion queue just like SM and SP commands.

*Brian

zemosh, 2016-08-04:
That would be great. 

Jiri

EmbeddedMan, 2016-08-10:
Jiri, a new version 1.4.1 now has been released. It has an updated SE command, where there is an extra optional parameter on the end. If the third parameter is a 1, then the SE command is added to the motion queue and will be treated just like an SM command.

Please give it a test and see if it works like you want it too. If not, let us know.

zemosh, 2016-08-10:
Brian, thanks a lot for the new version. I didn't expect such an early response. The SE command with the motion queue works exactly as I needed---from testing I did so far. Thanks again. I'm just curious:
1. What is the reason to set PWM to 50% if the power is explicitly set to 0? Is it because the power is 0 when no parameter is set? Then it would be better to state it in the documentation and allow only non-zero values otherwise it is a bit confusing---according to the documentation 0 is off.
2. Why is the power stored in a variable (StoredEngraverPower) instead of inside the CommandFIFO structure? If the FIFO was longer than 1 one day, it could cause losing some values.

-Jiri

EmbeddedMan, 2016-08-10:
Jiri,

Excellent questions!

1) It was sort of an arbitrary choice. I had to choose some value for the
power if the power parameter is missing. But if you explicitly set the
power value to 0, then the PWM percentage should be output as zero - is
that not the case? if it is not, I should fix that.

2) You are quite correct - if we ever expand the FIFO size to more than 1,
we'll need to add that power level into the struct put in the FIFO. I
should have done that when I made this change to future-proof the code.

*Brian

zemosh, 2016-08-10:
If I send SE,1,0, then the output is set to 50% so I'm using SE,0 instead.

-Jiri

EmbeddedMan, 2016-08-10:
Oops. That's a mistake. It should be 0% if you sent SE,1,0. I'll get that fixed. Sorry!

*Brian

EmbeddedMan, 2016-08-11:
Version 2.4.2 is now available and should fix this oversight. SE,1,0 now sets power to 0%. Also added SE power level to FIFO structure as you suggested.

closed 2016-08-11 by EmbeddedMan