evil-mad/EggBot #52: Parameters check in function parse_SM_packet (ebb.c) does not work propperly

zemosh, 2016-08-04:
The variable for the absolute value of steps (Steps) is shared by both axis (A1Steps and A2Steps)---axis 2 then overwrites axis 1. There should be two separate variables for the absolute values.

oskay, 2016-08-04:
@EmbeddedMan Can you please look into this?

EmbeddedMan, 2016-08-04:
Yup, looked into, and confirmed it is in fact a bug. I'll put a fix in shortly.

EmbeddedMan, 2016-08-10:
Version 1.4.1 now has this bug fixed. I used the following commands to test that the limit checking is working properly.

```
SM,20000000,100,100
!0 Err: <move_duration> larger than 16777215 ms.
SM,10000000,20000000,0
!0 Err: <axis1> larger than 16777215 steps.
SM,10000000,0,20000000
!0 Err: <axis2> larger than 16777215 steps.
SM,1000,26000,0
!0 Err: <axis1> step rate > 25K steps/second.
SM,1000,0,26000
!0 Err: <axis2> step rate > 25K steps/second.
SM,10000,7,0
!0 Err: <axis1> step rate < 1.31Hz.
SM,10000,0,7
!0 Err: <axis2> step rate < 1.31Hz.

```

Please test this zemosh with the new 1.4.1 firmware and let us know if you find any other problems.

closed 2016-08-10 by EmbeddedMan

zemosh, 2016-08-10:
I tested the new v1.4.1 firmware and the parameters check works now for both axis properly. Thanks for the fix.