evil-mad/EggBot #54: Clarify EBB command description syntax

oskay, 2016-08-10:
Following up on #51, I would like to request an update to the descriptions of individual commands on [the EBB commands page](http://www.schmalzhaus.com/EBB/EBBCommands.html). 

For reasons of clarity and usability, it would be nice to standardize the command (and query) description with slightly more rigid syntax.  While there isn't any one given standard for this kind of thing – Many different "standards" are used – there are certain norms and conventions that are common and that can be very clear when an explicit legend is given.

I would like to recommend that the following syntax legend (details of which are certainly up for debate!) prepend the command set, and that the commands be updated to follow this syntax.

------

EBB Syntax Conventions
* Command format descriptions and examples are set in `code font`.
* Query response format descriptions and examples are also set in `code font`.
* _Italics_ are used to represent variables and adjustable parameters.
* Square brackets (`[ ]`) are used to enclose optional items.
* Angle brackets (`< >`) are used to represent individual control characters, such as `<CR>` for carriage return.
* All unitalicized text and punctuation within a command description must be used literally.

Additionally, please note that:
* All commands are composed of ASCII characters
* All commands are case insensitive.
* Whitespace within a command will be ignored.
* All commands must have a total length of 64 characters or fewer, including the terminating `<CR>` .

------

To give an example, here is what the current description of the "SE" command looks like:

* **Format**: "SE,&lt;state&gt;,&lt;power>&lt;CR&gt;" (pre v2.4.1)
* **Format**: "SE,&lt;state&gt;,&lt;power&gt;,&lt;use_motion_queue&gt;&lt;CR&gt;" (v2.4.1 and on)
* **Example**: "SE,1,1023"
* &lt;state&gt; is 0 for disable, and 1 for enable. You must include this parameter.
* &lt;power&gt; is a value from 0 to 1023 and represents the relative power (% duty cycle of the PWM signal), where 0 is off and 1023 is full on. This parameter is optional.
* &lt;use_motion_queue&gt; is optional, and can be a 0 or a 1. If 1, then this SE command will be put on the motion queue just like SM and SP commands, thus it will be executed when the previous motion commands have finished. Note that if you need to use this parameter, the &lt;power&gt; parameter is not optional. (You must have 3 parameters if you use this parameter.)
* This command is used to enable and disable the engraver PWM output on RB3 (called B3 on the board), and also set its output power. Use SE,0 to disable this feature.
* If &lt;power&gt; is not included, 50% PWM will be used.
* Returns: "OK&lt;CR&gt;"
* Version Notes: Firmware versions prior to v2.2.2 do not have working PWM from this command. Instead, they always defaulted to 50% PWM. Version 2.2.2 fixes this problem.

Now, here is what that same description would look like with the new syntax (and a few edits):

* **Format**: <code>SE, <i>state</i> [, <i>power</i>]&lt;CR&gt;</code>  (Firmware < v2.4.1)
* **Format**: <code>SE, <i>state</i> [, <i>power</i> [,  <i>queue</i>   ] ]&lt;CR&gt;</code>  (Firmware versions &ge; v2.4.1)
* **Example**: <code>SE,1,1023&lt;CR&gt;</code>
* <i>state</i> is a required parameter, with allowed values 0 for disable, and 1 for enable. 
* <i>power</i> is an optional parameter, with allowed values 0 through 1023. It represents the relative power (% duty cycle of the PWM signal), where 0 is off and 1023 is full on. If <i>power</i> is not specified, the default value of 50% PWM will be used.
* <i>queue</i> is an optional parameter, with allowed values 0 or 1. If the value is 1, then this SE command will be put on the motion queue just like SM and SP commands, thus it will be executed when the previous motion commands have finished. If the value is 0, it will be executed immediately instead. Note that the power parameter must be specified if you choose to include this parameter.
* **Returns**: <code>OK&lt;CR&gt;</code>
* Version Notes: Firmware versions prior to v2.2.2 do not have a working PWM argument from this command, and always use 50% PWM. Firmware v2.4.1 adds the <i>queue</i> parameter.

----

A couple of notes on the example above:
* I've added the `<CR>` in the example.
* This set of changes makes it explicitly clear which items are parameters (for example, &lt;state&gt; is, but &lt;CR&gt; is not)
* This set of changes makes it explicitly clear which items are optional and which are not-- for example, the comma before <i>power</i> is required, if power is used, and that <i>queue</i> may be used only if <i>power</i> is. With the previous syntax, it is ambiguous as to whether the additional commas are optional or required.

A couple of additional thoughts in general:
* It might be nice to have some convention or explicit means of marking the the expected parameter values a little more clearly. One might consider the following means of doing that:
 * Adding to the command: <code>SE, <b>BOOLEAN</b> <i>state</i> [, <b>INTEGER</b> <i>power</i> [,  <b>BOOLEAN</b> <i>queue</i>   ] ]&lt;CR&gt;</code>  
 * Adding to the parameter as: "<i>state</i>: Boolean, required. Values: 0 for disable, and 1 for enable." 
 * Adding to the parameter, as "Boolean <i>state</i>: required parameter. Values: 0 for disable, and 1 for enable."
 * Using a convention where all boolean parameters are capitalized ("<i>STATE</i>").
* This is a mouthful; I'm open to discussing the types of formatting used, as well as helping out with the editing.

EmbeddedMan, 2016-08-10:
>    - *Format*: "SE,<state>,<power><CR>" (pre v2.4.1)
>    - *Format*: "SE,<state>,<power>,<use_motion_queue><CR>" (v2.4.1 and on)
>    - <state> is 0 for disable, and 1 for enable. You must include this
>    - <power> is a value from 0 to 1023 and represents the relative power
>    - <use_motion_queue> is optional, and can be a 0 or a 1. If 1, then
>    finished. Note that if you need to use this parameter, the <power>
>    - If <power> is not included, 50% PWM will be used.
>    - Returns: "OK<CR>"
>    - *Format*: SE, *state* [, *power*]<CR> (Firmware < v2.4.1)
>    - *Format*: SE, *state* [, *power* [, *queue* ] ]<CR> (Firmware
>    versions ≥ v2.4.1)
>    - *Example*: SE,1,1023<CR>
>    - *state* is a required parameter, with allowed values 0 for disable,
>    - *power* is an optional parameter, with allowed values 0 through
>    where 0 is off and 1023 is full on. If *power* is not specified, the
>    - *queue* is an optional parameter, with allowed values 0 or 1. If the
>    - *Returns*: OK<CR>
>    v2.4.1 adds the *queue* parameter.
>    parameters (for example, <state> is, but <CR> is not)
>    optional and which are not-- for example, the comma before *power* is
>    required, if power is used, and that *queue* may be used only if
>    *power* is. With the previous syntax, it is ambiguous as to whether
>       - Adding to the command: SE, *BOOLEAN* *state* [, *INTEGER* *power*
>       [, *BOOLEAN* *queue* ] ]<CR>
>       - Adding to the parameter as: "*state*: Boolean, required. Values:
>       - Adding to the parameter, as "Boolean *state*: required parameter.

Yeah! I love it. This is a fantastic idea, and something that's been sorely
needed from the beginning.

If you're willing to edit the commands file to match this new format, that
would be great.

Otherwise I can do it as I have some time (which might not be soon).

If you do the editing, the only thing I ask is that you send me any changes
to the commands HTML file that require changes to the firmware, or at least
file them as issues.

EmbeddedMan, 2016-08-10:
Would it work better to transition the EBB Commands page into the EBB Wiki?
Would that help in any way? (Anybody could edit it then, for example.)

oskay, 2016-08-10:
This is strictly a documentation request-- no firmware changes are anticipated here. (If any come up, we'll definitely start them out as independent issues. Each should be considered carefully.)

I'd be happy to work on the HTML... transitioning it to the wiki is a plausible method as well. Perhaps a github.io page would be the best compromise, and allow us both to edit, while not opening it up excessively.

Any comments on the syntax format ideas?

EmbeddedMan, 2016-08-10:
Out of the four suggestions you had, I like the first one best.

Overall, the whole reformat you are proposing looks really good. Makes
things look way more professional. Like in a textbook teaching you a new
language or something.

One thing: I don't think whitespace is ignored, but I need to check that.
We can of course change that if it would make a difference to anybody.

EmbeddedMan, 2016-08-10:
I just read up on github.io pages, and that seems like a perfect way to
host this content. I see that the EggBot repo doesn't have any GitHub Pages
set up yet. Is that something you'd like to do, and then I can add an EBB
Commands HTML page to it?

oskay, 2016-08-10:
Yes; let's proceed that way-- I'll get a shell set up shortly.  

Can you please check on the proper rule on whitespace?

zemosh, 2016-08-10:
It seems to me that no whitespaces should be use at all.

oskay, 2016-08-11:
I have started on this... Give me a little leeway to make some initial progress. 

Can you please specify which of the UBW commands do or do not work in the current EBB firmware?

EmbeddedMan, 2016-08-11:
Sweet. I won't do anything with it until you let me know you have something
ready to review. If you get part way and get sick of it, let me know and
I'll finish it up.

Whitespace is not ignored. There must be no extra spaces, tabs, etc. in the
commands or you will get errors. (This could be changed if it's a big deal
for anybody.)

Command names and parameters are not case sensitive.

UBW commands supported by EBB:

R, CU, T, C (although it is somewhat different from UBW version), AC, O, I,
V, A, MR, MW, PO, PI, PD, RC (but use S2 instead if possible), PC, PG, BL,

*Brian

oskay, 2016-08-11:
OK -- great. Just getting started.  

But, a preview: http://evil-mad.github.io/EggBot/EBBReleaseNotes.html

EmbeddedMan, 2016-08-11:
That is pretty nice.

My only suggestion is to have it scale to the width of the window. Right
now the 'content' part of the page takes up about %25 of the width of my
window, centered between two very wide gray bars.

*Brian

oskay, 2016-08-11:
You're welcome to look at the other github.io themes-- I tried several, and this one formatted the text the best. (I'll tweak the CSS a little -- 640 px is pretty damned narrow.)

One other item: A standard feature of protocols like these is the option to specify parameters as part of an exclusive-or set, as in the "EM" case, where the user needs to pick one of four items.  If we're already adding a new protocol convention that bold text can be used to describe variable types, then that makes this a little easier.

To the protocol conventions already mentioned, let's add the following:

* <B>Bold text</B> is descriptive information about parameters, such as the parameter type or name. This text must be <i>omitted</i> when using the command or query.
* Curly braces (`{ }`) are delimit a set of two or more items, separated by one or more vertical bars (`|`), of which exclusively one item must be chosen. 
* Vertical bars (`|`) are separate individual items between curly braces (`{ }`), of which exclusively one item must be chosen. 

Then, we can use syntax such as:

<b>Format:</b> <code>EM,  <b>Axis_1_State</b> {1|2|3|4}, <b>Axis_2_state</b> {1|2|3|4}&lt;CR&gt;</code>

oskay, 2016-08-11:
Can you please write a document for the "C" command on EBB?

zemosh, 2016-08-11:
- I wouldn't mix names in bold and italics. It could be confusing if you don't remember the legend exactly.
- Those exclusive-or sets can be described as integers with a range. It is, in fact, similar to parameters like power for SE (0--1023 is also exclusive set).
- I would leave types to the description. They are not as important here, and the syntax remains clearer. Most of the parameters are integers (Boolean is also in integer representation)
- It would be nice to use bold font for something that should stand out from the text, for example, commands (and all other constant text).
- (instead of <CR> it would be possible to use an escape sequence \r)

Of course, there is no perfect way how to do it and those are just my personal preferences, but I wanted to share them.

oskay, 2016-08-11:
* Please clarify what you mean by mixing bold and italic-- I don't think that we have any bold italic here.
* Yes, they are integers. However, this syntax is common in serial protocols where choices are limited to a specific set of values. It is also already in use in the command documentation, as in "SC,2,{0,1,2}" in the release notes.  Specifying this, to show that you must use a value of 0, 1, or 2 is compact and more clear than specifying "an integer in the range of 0 through 2."  
We are not using the syntax "0--1023" anywhere, to my knowledge. 
* I'm not dead set on adding the type specifications in the command description, but it is a common method, and the one that @EmbeddedMan seemed to prefer. I had suggested several means of putting it elsewhere as well – is there one of them that you disliked less?
* The commands will be set in the code font to distinguish them from the rest of the text on the page. To me, putting the "regular text" in the code font makes it look like code, and putting other text in some offset manner – such as bold – makes it look less like code. Another method would be to use a color coding (for example bold, but light gray) to indicate which text is for description only.
* Yes, escape sequences can be used as well, but perhaps the `<CR>` is more clear to distinguish that it is not a literal in the command description.  It might be reasonable to use `<CR>` in the command description, and `\r` in the set of examples for any given command.

zemosh, 2016-08-11:
- Names of required and optional parameters are in italics (e.g. state, and power), but names in descriptive information are in bold (e.g. Axis_1_State).
- The pipe convention is commonly used, and it's OK to use it. The problem is how to reference those parameters from the description. You have to use another typesetting for the name, and it's a bit confusing to me. An exclusive-or set usually needs some table in the description to describe the meaning of individual values, and this table is enough to understand that you have to pick one of these values.
- Type specification within the syntax is common in programming languages, but I've never seen it in serial terminal commands descriptions. I'm not telling it is not used, because my view is limited. I've tried to go through a few documents now, and types are either missing entirely or are stated within a table describing the parameters. I would put the types into the description (if somewhere) behind the name. Although it is common to have the type first in programming, here, I think, it's more important to have the name first in the description, for the sake of lucidity. 
- You are right with the bold font within a code environment. The light gray color of the bold text is a good idea---it would make it less highlighted.
- `<CR>` is OK and probably better than `\r`. I just thought it would be nice to have a possibility to transfer the syntax to plain text. For example, if you want to put it into comments inside a code. Then the angle brackets could be used to replace italics, and it would interfere with `<CR>`, but it is not a big deal.

Again, please, take this only as my personal point of view. It's up to you guys to decide. And thanks for your effort.

EmbeddedMan, 2016-08-11:
Sure. "C" command -

Here it is from the source code for EBB. We can expand on this as we
develop the updated command document.

// FORMAT: C,<portA_IO>,<portB_IO>,<portC_IO>,<portD_IO>,<portE_IO><CR>
// EXAMPLE: "C,255,0,4,0,0,0<CR>"
// <portX_IO> is the byte sent to the Data Direction (DDR, also called
TRIS) register for
// each port. A 1 in a bit location means input, a 0 means output.
//
// NOTE: it is up to the user to tell the proper port direction bits to be
// inputs for the analog channels they wish to use.

oskay, 2016-08-11:
Great. One more choice to make here: Should each of the inherited UBW commands be given a place in the (now neater, easier to read) EBB command set?

EmbeddedMan, 2016-08-11:
I would vote yes to that! Then users would only have to look in one place,
the EBB command doc, for everything that the EBB can do.

*Brian

oskay, 2016-10-16:
I'm starting to redirect some of our links to the new EBB docs; I would ask that you would do the same. (Adding additional UBW command links is still on the to-do list.)

EmbeddedMan, 2016-10-16:
Windell,

Yup, Thanks for the reminder. Will do-

*Brian

EmbeddedMan, 2016-11-10:
The new command docs are up, and include all UBW as well as EBB commands, as well as the new commands for the latest EBB firmware version 2.4.3.

One thing I noticed while reading the actual firmware source code is that every line that is sent back to the PC is terminated not just with \r but with \n\r. I will update the command document to reflect this.

EmbeddedMan, 2016-11-22:
The docs have been updated to show the proper bytes returned by the EBB at the end of each response line. I think this new document format is now fully ready for prime time, and this issue can be closed. Any problems with the command doc should be added as new issues.

closed 2016-11-22 by EmbeddedMan

oskay, 2016-11-22:
Awesome-- thank you. This is really a huge improvement. :D

oskay, 2016-11-22:
One final request on this: Please add a big fat link to the new docs at the top of http://www.schmalzhaus.com/EBB/EBBCommands.html

There are still a lot of links to your page out there. While I wouldn't advocate removing what is there, it would also be good to refer people to the new one.

EmbeddedMan, 2016-11-22:
Absolutely will do!

*Brian

EmbeddedMan, 2016-11-24:
And done.

*Brian

oskay, 2016-11-24:
Awesome-- thanks. Recommend word change of depreciated -> deprecated.

EmbeddedMan, 2016-11-24:
Yup. Agreed.

*Brain