evil-mad/EggBot #55: Is 10% speed increase possible "for free"?

ShelMi, 2016-08-14:
Eggbot.py uses the customary Euclidean distance measure to calculate the time required for each move.  Because of this, a move at delta x = delta y = n, is calculated to take about 41% longer than a move at e.g. delta x = 0, delta y = n.

"Of course" you say - the distance is 41% greater!  But wait...each stepper is still only moving n steps!!
The increased time resulted in _decreasing the stepping rate_ of each of the two orthogonal stepper axes.  Thus a movement to a point at 45 degrees will require have the stepper motors moving at only about 70.7% of the user-specified stepping rate.

Since the two axes of movement are orthogonal, could not each stepper move independently at its user-specified steps per second?  If this were true, the the proper distance measure would be
```max( abs( delta x ), abs( delta y) )``` rather than ```( x^2 + y^2 )^.5```
I think the proper name for this metric is a "maximum metric" - it's vaguely like an inverse taxicab metric.

If we assume that a typical plot has a relatively uniform distribution of movement angles, then the plotting time improvement from changing to this metric would be about ( pi/4 - sin(pi/4) ) / ( pi/4 ) or about 10%.

I readily acknowledge that I am not cognizant of all the factors that went into the design decisions,
but I am just wondering out loud if this distance measure is perhaps a reflex of our conventional Euclidean impulses, rather than a deliberate decision.

oskay, 2016-08-14:
I can see where this comes from. I suppose the question is whether the speed rate there is to set the maximum speed at which the _steppers_ turn, or to set a speed that the _pen_ should travel at.  The present method gives "constant" pen speed, regardless of the angle that a line is being drawn.

There are some use cases where this does potentially make a difference. One could certainly argue that it would be optimal, for example with the diamond engraving tool or electro-kista, to have a uniform drawing speed. On the other hand, the "constant" pen speed is anything but, right now, because of projection mapping-- especially towards the poles of the object.  It does make pen lines have uniform thickness near the equator, though.

Perhaps the optimal solution would be to try to optimize the pen speed to be constant over the entire surface, compensating for the fact that more steps are required at the poles.

ShelMi, 2016-08-15:
I get it now - thanks for broadening my perspective on the eggbot!

I'll close the issue tomorrow, unless you want it left open as a reminder of the possibility of addressing the variability of the "constant" pen speed - particularly approaching the poles as it affects engraving and wax deposition.

oskay, 2016-08-15:
Yes; please leave it open. It's good food for thought on future speeds. I'll edit the title to reflect.

oskay, 2016-08-15:
One possible UI interface for this type of thing: We could use the "Enable engraver" selection box to enable the constant-drawing-speed mode, potentially with correction for the map projection.

ShelMi, 2016-08-15:
Oooh, that sounds like the _perfect_ UI method!