evil-mad/EggBot #61: EBB firmware: Query Stepper position

oskay, 2016-11-07:
Feature suggestions for future EBB firmware:
Query the EBB; report its current position with respect to the initial starting point.

<b>Command:</b> <code>QS&lt;CR&gt;</code>  
<b>Response:</b> <code>_x_position_,_y_position_&lt;CR&gt;OK&lt;CR&gt;</code>

EmbeddedMan, 2016-11-07:
Hey, that's a really good idea! Should be very simple too. Note that it will only be accurate if there is no motion currently happening.

oskay, 2016-11-08:
It will be necessary to make that _other_ feature request worthwhile. 
(In retrospect, I'm kind of floored that we don't have this already.)

EmbeddedMan, 2016-11-08:
Agreed. This just seems to make good sense.

We should also have a "CP" for Clear Position to set both x and y positions
to 0 at any point.

oskay, 2016-11-08:
How about CS and QS for clear step count and query step count?

EmbeddedMan, 2016-11-08:
That would work just fine.

EmbeddedMan, 2016-11-08:
Created v2.4.3 EBB firmware release and committed it to the repo. Added QS and CS commands. In my testing, both work as intended. Also updated documentation and release notes html files.

Please test, and close this issue if they meet your needs.

EmbeddedMan, 2016-11-09:
> Thanks for getting this done instantly (whoah!). I haven't had a chance to
> test this yet.
>
> At what points is the position automatically set to zero? I can imagine
> setting it to zero:
>
>    - At reset
>    - When a stepper motor is energized (with the EM command).
>    - When a stepper motor is de-energized (with the EM command).
>    - When a step resolution is changed (with the EM command) -- assuming
>    that each step is considered "a step" with respect to the step counter (as
>    opposed to counting a full step as 16 steps, a half-step as 8 steps, and so
>    forth).
>
> Or in other words, at reset and any time that the EM command is executed
> such that it makes a change.

The position is only reset to zero at boot and when you send the CP
command. It feels to me like letting the PC manage when it gets reset is
more generic and flexible than forcing it to be reset automatically at
other times. The PC can simply send the CP command at any of the events you
suggest (like the EM command) if it wants to clear the positions.

*Brian

EmbeddedMan, 2016-11-15:
Have you had a chance to test this command out at all? Are you satisfied with how it works?

oskay, 2016-11-15:
I have not yet had a chance to test it.

ertdfgcvb, 2017-09-06:
I tested both the `QS` and `CS` commands and they work well. Thank you!

Two notes: 
1.
For consistency it would be appropriate for ALL query commands (`Qx`) to prepend the query command in the response… currently only `QM` does this. 
This could be very useful for a generic / realtime message parser…

2.
The `QS` query return values correspond to the values set by `SM`. 
Would it be appropriate to have a `QX` query when using `XM`? 
Currently we need to convert manually: `x_steps = x_query + y_query` and `y_steps = x_query - y_query`… not a big deal, but could be a bit confusing.

closed 2019-01-09 by oskay