evil-mad/EggBot #62: EBB firmware: Return home on holding button

oskay, 2016-11-07:
Feature suggestions for future EBB firmware: Hold pause button to return home. 

Presuming that the EBB can keep track of the number of steps taken since the last time that it was energized, holding the PRG (pause) button for three seconds could cause the EBB to drive its motors back to the home position, either at some default velocity, or at "typical" velocity set in firmware.

ShelMi, 2017-04-19:
I wonder if a possible problem with this feature might occur in the following scenario?

   User presses pause button, but eggbot is in the midst of a long straight move.  (By "long", I mean one which will take longer than three seconds.)

User sees that movement did not cease immediately, and *doesn't realize that EBB has seen the button* and will pause at the end of the move.   User continues to hold down the pause button, mentally repeating "Stop goldarn it, Stop goldarn it, why don't you stop goldarn it..." until the end of the move finally occurs.

User has by now inadvertently exceeded the three second "return to home" duration, resulting in an unfortunate inability to resume plotting.

oskay, 2017-04-19:
Yes, that is a potential issue.

And yes, pressing it would cause a failure of our current pause-resume system.

Perhaps it would need to be accompanied by a corresponding set of enable/disable commands that would make it so that this command could not actually be called while it is plotting.

EmbeddedMan, 2017-04-19:
Or maybe it could be simpler than that - a long-press would be ignored
unless there are no motion commands in the queue?

What's the use case for the long button press return to home feature? Would
we only want to trigger it after a plot is complete and the system is idle?

*Brian

oskay, 2017-04-19:
I suspect that querying the queue would not solve the issue-- if you're in the middle of a long move, you have no way (as the human pushing the button) to know how much is left in the queue. 

A secondary tricky part is that all EBB-using software would need to be aware of this feature (at some level)-- ideally querying the step counters to keep track of current position.

Use case is general setup-- suppose that you pause a plot (for fixturing/settings changes) and need to do so at the home corner. This avoids having to request that from within software-- it's a natural user interface for it, and would save the user time versus going back to the computer.

EmbeddedMan, 2017-04-19:
Sorry, what I meant was "A long-press would be ignored (by the EBB) unless
there are no motion commands in the queue and the motors are not currently
executing a motion command". In other words, the EBB would not even
register that a long press happened unless the system is idle. There would
be no need for the PC to query the queue at all in this case.

You are absolutely right that if this feature gets implemented, we should
have it disabled by default, and give the PC software the chance to enable
it. That way existing software doesn't need to change, but future software
could enable the feature if it knows how to query the step counters.

*Brian