evil-mad/EggBot #69: Think about the XM command...

oskay, 2016-12-02:
We have been discussing the benefits of the XM vs SM command, here: 
https://github.com/evil-mad/EggBot/issues/63#issuecomment-264529402

Let's talk: Does it make sense for the future to changing how the inputs are checked? For example, to have a different (possibly more restrictive) check on the inputs of the XM command, but one that allows the entire computation to be done in the (A,B) space?

EmbeddedMan, 2016-12-02:
> #63 (comment)

Honestly, this is probably what the XM command should have been from the
beginning. It's got to have some reason for existence - some reason why the
PC app would be better off using it than SM.

On the other hand, all 'host' systems (PCs, tablets, Pis, etc.) will have
far more processing power than the little EBB, so if the _only_ reason the
XM command exists is to have the EBB do some math rather than the host do
the math, maybe that isn't a good enough reason.

*Brian

oskay, 2016-12-02:
The reason for it to exist is to be intuitive. When using machines with mixed-axis geometries, it is helpful if the end user never has to think about "A+B" -- they only need to think in X and Y.

EmbeddedMan, 2016-12-02:
OK, so what kinds of limits should be placed on its parameters? Should we
try to make it as similar to SM as possible in terms of limits?

oskay, 2017-03-18:
Yes, it should have some kind of clearly defined criteria, such that one does not need to convert back to X and Y to check the limits. I'm not sure exactly what the most natural way to provide these limits are.

pacvis, 2019-04-20:
So is this a thing?

EmbeddedMan, 2019-04-20:
Yup: http://evil-mad.github.io/EggBot/ebb.html#XM

oskay, 2023-04-28:
Closing issue due to a lack of direction.

closed 2023-04-28 by oskay