evil-mad/EggBot #71: Gap in time between certain back to  back SM commands

EmbeddedMan, 2017-01-07:
The EBB firmware (tested v2.4.3, assumed same for all previous versions) adds little gaps between some back to back SM commands, which really puts a damper on nice smooth motion. 

For example:
SM,500,2000,50
SM,500,2000,50
Will produce 1 second of continuous step pulses.

But 
SM,500,2000,1
SM,500,2000,1
will produce 500ms of step pulses, a 147ms gap with no step pulses, and then 500ms of step pulses.

This is clearly wrong, and is likely due to a math bug in the EBB firmware.

EmbeddedMan, 2017-01-07:
Here is what the bug looks like:
SM,500,2000,1
SM,500,2000,1

[image]

The interesting thing of note here is that the 1 step for the 2nd axis does not happen at the end of the 500ms. Rather, it appears that the 2000 steps of Axis 1 take 500ms, then the 1 step of Axis2 takes 147ms.

oskay, 2017-01-07:
Similar behavior can be found with other movements of high aspect ratio, for example
`SM,500,2500,10`
`SM,500,2500,10`

EmbeddedMan, 2017-01-07:
I think the effect is caused by integer math inaccuracies when computing the time between steps for small numbers of steps. 

For example, if you do
SM,1000,2,0
you'll see that there are actually 660ms between the two steps - there should be exactly 500ms. 

It turns out that we really need 64 bit integer math (or, better yet, floating point) in order to get accurate values in the calculation of the value we add to the accumulator each 40us, and the (old) C compiler we're stuck with for the moment doesn't have any 64 bit libraries. There is one big division which , when done as a 32 bit number, results in significant loss of precision. The effect is worst when the number of steps per second is small I believe.

The good news is that I think I've found a way to redo the math operations, using some 32 bit remainder operations, to fix this issue. I'm testing this change now.

oskay, 2017-01-07:
Perhaps it would be possible to pull in a special case to manage the division separately when one axis is (say) < 20 steps.

EmbeddedMan, 2017-01-07:
Unfortunately, the error is always there, no matter what the values are.
It's just easiest to observe in human-time-scale with high aspect ratios
like you've discovered. To me it's better to fix the fundamental problem,
and have it work for all values. I've already got the fix in place, and it
is working. I'm now trying to test it out in a number of ways and
hand-check the results.

EmbeddedMan, 2017-01-07:
I have a new version, which has the bug fixed.  It is version 2.4.5. Please download it to one of your EBBs and test it out with AxiDraw to see if everything still works OK and if you can confirm that the problem is now gone.

This Logic trace shows that there is no longer any gap. (same commands as above)
[image]

oskay, 2017-01-07:
Yeah, wow. Holy crap that is better!

EmbeddedMan, 2017-01-07:
Great to hear. I'll close the bug, and create a Windows auto-installer for
the new version. If you find any problems with the new version, file a bug.

oskay, 2017-01-07:
The high-precision plot is finishing now, and doesn't show any signs of trouble. I think that we likely have a winner here, although this does not cover all possible working conditions.

Great. It looks like I have a reason to build that Mac auto-installer now, too.

EmbeddedMan, 2017-01-08:
The updater application (for Windows) has been pushed.

closed 2017-01-08 by EmbeddedMan

EmbeddedMan, 2017-01-08:
So the one downside of this change is that the shortest move that the EBB can make without any breaks between moves went from 15ms (with v2.4.4) to 20ms (with v2.5.5). So if you try to execute a sequence of moves where the move durations are less than 20ms, you will have small gaps between your moves.

This is because of the heavy lifting math (we went from 2 32/32 bit divides to 4 in the new version) that has to happen to convert the duration and step counts into the proper internal values that get placed in the movement FIFO. This math happens once per command.

Here's what version 2.4.4 looks like with the following commands:
SM,15,15,15
SM,15,15,15
SM,15,15,15
SM,15,15,15
SM,15,15,15
SM,15,15,15

[image]

And here's version 2.4.5. with the same 15ms moves:
[image]

And here's 2.4.5 with 20ms moves
SM,20,20,20
SM,20,20,20
SM,20,20,20
SM,20,20,20
SM,20,20,20
SM,20,20,20

[image]

oskay, 2017-01-08:
Three things: 
1. I've got an updater Application for Mac almost ready.
2. A lot of the EggBot and AxiDraw movements do take place at shorter time scales. Since the duration is explicitly requested each time, would it be possible to put in a special case to handle short moves, say under 20 ms? It might make it possible to get to a shorter-yet minimum move durations.
3. If there is a known gap between moves, it should be explicitly stated under the `SM` command documentation.

EmbeddedMan, 2017-01-08:
We could certainly add special case code to not do the new divisions for moves less than 20ms. That would then allow you go back down to 15ms moves like in previous versions. Is it common enough for you to have sub 20ms moves that this change would be worth it? Even with the change, you won't be able to safely go below 15ms without introducing gaps.

I will add a note to the docs.

oskay, 2017-01-08:
There has never been a special case to handle shorter moves-- we have always allowed moves down to 1 ms in duration in the software... which probably means that he behavior around tight curves would be better if shorter moves were available.

EmbeddedMan, 2017-01-08:
I have no doubt. However, at this point, we would have to get rid of some features in the EBB firmware to allow the little 12Mhz 8-bit processor to compute moves in less time, thus allowing faster back to back short moves.

A much easier way is simply to have the PC do all of this math (it's 2 64 bit divisions) and add a new command that would have the PC simply send the pre-computed values to the EBB. Then the EBB would simply receive the values and stuff them right away into the FIFO with no math. This would be fastest, and much easier to implement in the firmware. If this is something you want to do, add a feature request and we can discuss it there.

oskay, 2017-01-08:
Not quite sure what that would look like, but I'll open a separate issue for it.

Please do consider adding the special case, at least, so that this will not add new "gaps" in those 15-20 ms cases.

EmbeddedMan, 2017-01-08:
I have created a new version, v2.4.6, with the optimization. For all moves less than 30ms in duration, we skip the extra precision code added in v2.4.5. This now allows v2.4.6 to have 16ms back-to-back SM commands with zero gaps. (15ms is usually OK, but every now and then (depending on how USB is feeling that millisecond) you'll get a little gap. 16ms seems to work all the time.)

Would you mind testing out v2.4.6 on your difficult path to make sure no bugs were introduced by this change?

oskay, 2017-01-08:
Will do, thanks.

EmbeddedMan, 2017-01-08:
I started a new RM command issue  https://github.com/evil-mad/EggBot/issues/72

oskay, 2017-01-08:
I did say that I would, but it took me a few minutes longer to type it up.

EmbeddedMan, 2017-01-08:
Yeah, no worries at all Windell.

oskay, 2017-01-09:
Testing appears to be successful with 2.4.6. I've also added the mac application to update firmware to the repository.

EmbeddedMan, 2017-01-09:
Great! I've added a 2.4.6 updater app for Windows. And actually checked in
the exe file this time. :-|