evil-mad/EggBot #72: Add a 'raw move' command to EBB firmware to allow fastest possible moves

EmbeddedMan, 2017-01-08:
Right now, the SM and XM commands in the EBB are what move the motor. You send in a move duration, and two step counts (one for axis 1, the other for axis 2). Internally, the EBB firmware then converts these three numbers into a set of values which are used by the motion control system to deliver the step pulses to the steppers.

This conversion requires a few math operations that take a long time on the EBB (because it's processor is a 12MHz 8-bit unit), and thus pose a lower limit on the length of move commands that can be processed back to back without any gaps in the step pulses.

Currently (in v2.4.6) this lower limit is about 16ms. If multiple moves with shorter times than this are send to the EBB, there will begin to  be small gaps between the moves' step pulses because it takes about 16ms to do the step processing.

One solution is to create a new command, called RM (raw move). It will allow the PC application to do _all_ of the math required, and just send the raw values needed by the motion system to the EBB. Thus, no processing will have to be done on the EBB, it can just push the raw values right into the motion control FIFO. 

This will result in the shortest minimum move duration possible with the EBB.

oskay, 2017-01-08:
"duplicate of #73"

oskay, 2017-01-14:
Closing this in favor of #73. (If you'd prefer to close the other and re-open this, that's fine. But one is enough.)

closed 2017-01-14 by oskay