evil-mad/EggBot #74: Feature request: Save timing choices in .svg file

ShelMi, 2017-03-16:
I recently did hours of experimental hill-climbing to optimize settings using the engraving tool.  Foolishly I did not make a note of my final choices.  Had there been an easy/quick way to save my timing choices in the .svg file, I probably wouldn't be kicking my non-note-taking-self now.

Then, Eureka!, I see that a related need is now being met in the AxiDraw driver, where printing data is being stored in the layer names.  What a clever place to store data, Zowee!

I have made an implementation adapted for Eggbot:  An added choice on the Eggbot "Manual" tab automatically makes a new layer whose name encodes the current "Timing" tab values in easily understood text.  The plotting software, if it finds such a layer, uses these values as overrides of the tab values.  The clear-language nature of the name makes it readily editable by the user should he/she want to.  This does not have the flexibility of the AxiDraw's per-layer specification, but it does have the virtue of being a swift non-technical way of saving adjustment progress.

In case there is any interest in seeing my whack at it, attached images are the message one gets when selecting "Save timing values into a new layer" from the Manual tab, and of a typical layer name.  Also attached is a zip of eggbot.py and .inx

There may well be better ways to implement this than the method I've tried, but in any event some version of this feature would probably be very useful to experimenters such as I.

[image]

[image]

[eggbot-2_8_2.zip](https://github.com/evil-mad/EggBot/files/847587/eggbot-2_8_2.zip)

oskay, 2017-03-16:
I can see what you are doing, and the general idea of it does make sense, but for several reasons I would like to steer this in a different direction. 

(1) My long-term goal is to merge much of the motion control "engine" into a library (part of plotink) that can consistent between EggBot, WaterColorBot, and AxiDraw. Features added to one usually come back to join the others eventually. In keeping with that approach, I'd prefer to use the same syntax for by-layer control. We can move that feature over from the AxiDraw code now, if you'd like it. (That implementation is currently documented here:  http://wiki.evilmadscientist.com/AxiDraw_Layer_Control )

(2) It makes sense to have per-layer controls in the layer names, but I don't think that it is as appropriate to have other types of controls (for example these, which are settings for all layers).

(3) Layer names can get really long this way-- too long to read.

Let me instead suggest that a better way forward would be to create a _documentation layer_ that contains the settings. This layer could be called (for example) "%EggBot Settings%", and would be hidden by default.  Since it has the "%" prefix, it would be a documentation layer that does not print. 

On that layer, you could create a flowed text box, starting at the upper left corner of the page, that contains all of the settings values that you would like to save, spelled out in plain text, and user editable, should the user choose to un-hide that layer.  It would be a good idea to give the layer and textbox each a specific id (e.g., `id="SettingsLayer"` `id="EggBotSettings"`) so that you can easily find and modify their respective contents easily

This layer could still be added through a GUI-based command (yes, possibly on the manual command list), and any additional notes or documentation written on the layer would also not be printed.

It should also be the case that:
* Per-layer overrides should have highest authority
* In the absence of a per-layer override, values from the Settings layer should be used
* In the absence of a settings layer, values from the GUI should be used.

ShelMi, 2017-03-23:
As usual, a very clarifying and edifying response - thank you!  Plus, it has given me another chance to indulge in the joyful process of learning/programming.

Attached is a version which addresses your concerns.  I call it "1-eggbot" rather than "eggbot", so that you may try it out without disrupting your extensions directory too much.

I think your suggestions yielded a rather nice implementation.  (Then again, I often think too highly of my current program versions.)  Anyway, I've enjoyed working on it and hope you get a chance to try it.  Further refinements/comments appreciated.
[1 - eggbot.zip](https://github.com/evil-mad/EggBot/files/863435/1.-.eggbot.zip)

ShelMi, 2017-04-12:
A couple of refinements, the need for which was found during actual use:
1. Allows, but does not require, user to maintain multiple settings layers, distinguishable by suffix.
2. Added space characters to the left of the values text, to make it easier for user to get the cursor into the proper text.
3. Added tooltips to describe how storing the settings works.

[image]

[1 - eggbot (2).zip](https://github.com/evil-mad/EggBot/files/912693/1.-.eggbot.2.zip)

oskay, 2017-04-12:
I apologize that I haven't had a chance to review this thoroughly yet-- this looks like a good addition, worthy of merging into the main trunk. I too will need to spend some time trying it out and seeing what I can do to make sure that it works well and is not too obtrusive.

ShelMi, 2017-11-23:
[1 - eggbot 2_8_2.zip](https://github.com/evil-mad/EggBot/files/1497934/1.-.eggbot.2_8_2.zip)

I've prepared a version which refines the operations of the Timing tab, as well as a couple of other goodies.  I have not implemented your single-layer-override schema, as it would need to be properly integrated with your broader code suite.

Testing has been limited.  A couple of moderately complex (10 or 20 minute) plots tested out to about 1% accuracy on the timing prediction - calculation time was four to eight seconds.  Helps answer that crucial question: "Will I have time to make coffee while the plot runs???"

Here are the more notable changes from your version 2.8.1, though a diff will show a few more minor changes:
```
# Notable changes from version 2.8.1:
# 1. Timing settings can now be stored as editable text in a documentation layer in the svg file.
#    This may be useful for saving the particular timing settings needed by individual files.
#    (See Note 1 for more details.)
# 2. Adds an information text box when 'Apply' the Timing settings.
#    It calculates the expected total plot time, as well as the amount of plotting time spent
#    in each of the activities (pen up delay, etc.) affected by the settings chosen.
#    (May prove useful for optimizing settings, as well as predicting
#    how long a coffee break to take during plotting.)
# 3. When "Apply" on timing tab, if delay settings so are so small
#    that pen motors move before the pen servo stops, generates an information popup.
#    This could reduce the chance of user inadvertently moving steppers before servo is finished.
#   (This is only needed while delay start time is at the start of servo motion, rather than at the end.
# 4. Adds check motor voltage command to the "Manual" tab.
#  
# Note 1:
#   a. A checkbox selects whether or not to store the current settings in a documentation layer.
#   b. A checkbox selects either removing earlier settings documentation layers,
#      or simply appending the latest layer while adding a sequence number to already-existing layers.
#   c. A checkbox selects whether to use the settings on the Timing tab, or
#      to use the values in the (editable) text of the documentation layer.
#   d. Allows user to maintain multiple settings layers, identifiable by suffix.
#      e.g. named as:
#        %Eggbot Settings% for engraving
#        %Eggbot Settings% for Sharpie xmas balls
#        %Eggbot Settings% test A
#      Only the non-suffixed name %Eggbot Settings% is picked as the source for using preserved settings.
```

ShelMi, 2017-12-27:
Have added some more information to the output:
[image]