evil-mad/EggBot #75: Feature request: Save hatch fill choices in .svg file

ShelMi, 2017-04-11:
Sometimes I have a pattern which I want to use on different ball sizes or surfaces.  Each combination may require rather differing settings for hatch fill.  It would be convenient to have a way of having some standardized, 'tried & true' values available in the svg file.

So, I've made changes to eggbot hatch to allow for hatch fill settings to be preserved in layers, similarly to the method of issue #74.  This is in the attached zip, and is named '1 - eggbot hatch' for ease of trying it out without disturbing existing version.

A user might reasonably add several of these settings layers to the default template, so they will be readily available when filling - rather than head-scratching and re-calculating particular pen width on particular destination surfaces.

[image]
[1 - eggbot_hatch.zip](https://github.com/evil-mad/EggBot/files/912716/1.-.eggbot_hatch.zip)

oskay, 2017-04-12:
Interesting-- it's a little harder for me to see that this kind of per-document settings set is absolutely needed within the hatch extension; I haven't really had a need to regularly adjust more than one or two parameters.

ShelMi, 2017-04-13:
I see what you mean - I rarely use more than two or three.

The two that I always use are hatch spacing and inset from edges - to minimize plot time I have to take into account both the size of the "egg" and the pen tip-width of the pens I'm using.

   The way I would use this feature would be to have a multitude of '%Hatch Settings%' layers, each with a descriptive suffix, e.g. 'peewee egg copic 0.2', or '80mm ball sharpie fine', or '45 mm ball paint pen'.  Then I would duplicate the layer which describes my current plotting situation, and remove the suffix from the duplicate.

I sometimes use crosshatch, depending on whether I feel a need for maximum ink laydown.

I occasionally use hatch angle if there is a dominant angle I would like to match.