evil-mad/EggBot #77: EBB firmware: Inconsistencies in query whitespace

oskay, 2017-06-10:
According to our documentation, the QB and QP commands should both return:

`state<NL><CR>OK<NL><CR>`
`state<NL><CR>OK<NL><CR>`

However, this is not actually what they return. 

QB returns
```
0^M
OK^M
```

while QP returns
```
0
^MOK^M
```

There may be additional issues of this nature.

EmbeddedMan, 2017-06-10:
Whoops. I thought I had standardized all of the line endings. Guess not. Easy fix.

oskay, 2017-06-10:
The tricky question is _should we_ -- changes like these could potentially break software that relies upon them.

EmbeddedMan, 2017-06-10:
True. How much software uses QB or QP?

Maybe the best solution, under the circumstances, is simply to update the
documentation.

oskay, 2017-06-10:
>True. How much software uses QB or QP?

Other than EggBot, AxiDraw, and WaterColorBot, I have no idea. :P

oskay, 2017-06-10:
It would also be a good idea to do a thorough check of all the other queries to see if they are correctly documented. 

If the "new version" firmware ever ends up happening, it would be nice to standardize all of the queries on the same response syntax. Right now we check for a second response line on everything except {"v","i","a", "mr","pi","qm"}.

EmbeddedMan, 2017-06-10:
Absolutely agreed. I'll take care of that right now.

And yes - if there's a new 'big' version of the firmware where we can make
a break from the existing lineage, we'll want to clean up a few things,
including this.

EmbeddedMan, 2017-06-10:
The good news is that, after reviewing the code, there don't appear to be any lone \r or lone \ns. Every line is terminated by either \n\r or \r\n.

It turns out that the fundamental problem here is that the linefeed sequence after the "OK" is always \r\n. For every command in the documentation, we said it was \n\r, so I've gone and fixed the documentation. 

I've also gone through all the rest of the printed strings in the code, and updated the docs so that it matches what's in the code for every command.

See the diff if you want to see what we got wrong with the previous docs: https://github.com/evil-mad/EggBot/commit/d66509c97b56b4b0424929a583925dcee8e61575

oskay, 2017-06-11:
That was _a lot_ of changes. Good.

closed 2017-06-11 by oskay