evil-mad/EggBot #78: LM command: Detect and reject "0,0" case

oskay, 2017-06-15:
Looking through the LM command, I was struck that it's relatively easy to get it into a state where you send it a command to do nothing and it effectively crashes the EBB into an infinite loop.

While I understand and appreciate that thorough checking of the inputs and their consequences is likely best kept out of scope for the LM command, this particular failure mode is nonintuitive and easy to trigger. That, combined with the fact that the mode is relatively easy to detect (and thus low on overhead) -- suggests that it would be a solid improvement to trap this one case. 

If I understand correctly, the solution would be to add something like the following to the parse_LM_packet(): 

```
        if ( ( (StepsCounter1 == 0) || (StepAdd1 == 0) ) && ( (StepsCounter2 == 0) || (StepAdd2 == 0) ) ){
                  return;
        }
```

Then, it would behave exactly as expected-- you ask it to move zero steps, and it complies (by not executing a move) without error or crashing.

EmbeddedMan, 2017-06-15:
Yup, good point. This will be super easy to take care of. Crashes are not
good.

EmbeddedMan, 2017-06-17:
I've created version 2.5.2 with a fix for this issue. Please give it a try and see if it works for you.

oskay, 2017-06-17:
Thanks-- I'll take a look.

oskay, 2017-06-19:
The change that you made adds:
```
    if (
        ((StepsCounter1 == 0) || (StepAdd1 == 0))
        &&
        ((StepsCounter2 == 0) || (StepAdd2 == 0))
    )
    {
        printf((far rom char *)"!0 Err: Invalid parameter set detected.\n\r");
        return;
    }
```

I would prefer to simply return in this case, rather than throwing an error. I expect the most common case to be simply asking for (0,0) steps-- and I think that "expected behavior" in that case would be for it to comply by moving zero steps and return without complaining.

EmbeddedMan, 2017-06-19:
Ahh! That makes sense. I'll update the version without the error.

*Brian

EmbeddedMan, 2017-06-19:
OK, new binary for v2.5.2 of EBB Firmware now available without the error
message.

oskay, 2017-06-20:
Great-- thanks!

oskay, 2017-07-03:
I think that this still may not quite be right. My understanding (now) is that the actual conditions to worry about are:

If ((StepAddi == 0 and StepAddInci == 0) OR (StepsCounteri == 0)) for both axes

Case in point: The following two moves appear to be valid, working moves under v2.5.1: 

`LM,17320000,1000,0,0,0,0\r`

`LM,0,1000,173200,0,0,0\r`

EmbeddedMan, 2017-07-03:
OK, should be a very simple fix. Do you want a new version number, or
should we just reuse 2.5.2?

*Brian

oskay, 2017-07-03:
Please recycle v2.5.2.

oskay, 2017-07-03:
Related note on LM documentation: I think that example 6 is not correct -- it exceeds 25 kHz step rate by the end.

EmbeddedMan, 2017-07-04:
OK, does this logic look right?

```
    /* Quickly eliminate obvious invalid parameter combinations,
     * like LM,0,0,0,0,0,0. Or LM,17320000,10000,0,0,0,0 or 
     * LM,0,1000,173200,0,0,0 GH issue #78 */
    if (
        ((StepAdd1 == 0) && (StepAddInc1 == 0))
        ||
        (StepsCounter1 == 0)
        ||
        ((StepAdd2 == 0) && (StepAddInc2 == 0))
        ||
        (StepsCounter2 == 0)
    )
    {
        return;
    }

```

EmbeddedMan, 2017-07-04:
Nope, that logic is wrong. Fixed to be:

```
    /* Quickly eliminate obvious invalid parameter combinations,
     * like LM,0,0,0,0,0,0. Or LM,17320000,10000,0,0,0,0 or 
     * LM,0,1000,173200,0,0,0 GH issue #78 */
    if (
        (
            ((StepAdd1 == 0) && (StepAddInc1 == 0))
            ||
            (StepsCounter1 == 0)
        )
        &&
        (
            ((StepAdd2 == 0) && (StepAddInc2 == 0))
            ||
            (StepsCounter2 == 0)
        )
    )
    {
        return;
    }

```

EmbeddedMan, 2017-07-04:
Also about example 6:
`LM,42949573,50,25769804,0,0,0`

If I run this line, it appears to do what the comments say : it starts off at about 500Hz step rate, ends at about 806 Hz, and lasts for about 76ms. This was measured on my logic analyzer. By the end, it does not get close to the 25KHz max step rate.

Now, what's interesting is that your example
`LM,42950000,50,13400,0,0,0`
ALSO does EXACTLY the same thing. :-) 

So, I need to dig in a bit to figure out why. Mathematically, your example is the correct one. So why is the original one not running way too fast? Digging into it now.

EmbeddedMan, 2017-07-04:
Ahhhh - found it. In parse_LM() we read in the DeltaR (StepAddInc) values as 32 bit numbers, but the structure that holds them through the move fifo only uses 16 bit numbers. So the 25769804 from the example actually gets stored as 14156, which is very close to your 13400. 

So, I will update the example and the text for the LM command to reflect that DeltaRs are only treated as 16 bit numbers.

However, we should think - does that limit the LM command (in a practical way)? Do we want to treat DeltaR as a 24 or 32 bit number? Or is 16 bit signed good enough for what we need it for right now?

EmbeddedMan, 2017-07-04:
OK, new v2.5.2 firmware is up and LM docs corrected.

oskay, 2017-07-05:
That zero-movement logic looks correct. 

As for the 16-bit limit on delta... well.... that... would explain some things that weren't making sense.

And, I don't think that it's enough to be useful for what we were trying to do.

Suppose we start at a step rate of (say) 1000 Hz. Then, the initial RateTerm is R0 = 85,899,345.
Suppose we want to accelerate as fast as possible from that speed to the maximum step rate, 25 kHz. (We can stop just short of a full 25 kHz, of course.)

Set DeltaR to its maximum, DeltaR = 32767. Then, the value of the RateTerm after _n_ 40 us time ticks is:

_R_ = R0 + (_n_-1) DeltaR.

Maximum speed occurs when R = 2^31, or 2,147,483,648.  So, _n_ = ((2^31 - R0)/DeltaR) + 1 = 62917.
 
That is to say, it takes over 2.5 seconds to ramp up to maximum speed. And, that's a lot slower than we can accelerate by using discrete movements with SM. 

If DeltaR were allowed to be a 24-bit signed number, then we could have DeltaR as large as 8,388,607, which would give us _n_ = ((2^31 - R0)/DeltaR) + 1 = 246, about 10 ms -- way faster than the mechanics could keep up.

EmbeddedMan, 2017-07-05:
Yup, you''re right. No doubt, we need larger DeltaRs. OK, I'll get that in.

EmbeddedMan, 2017-07-07:
OK, a new v2.5.2 is now up, with 32 bit DeltaR parameters. I tested the example above
`LM,42949573,50,25769804,0,0,0`
and it does exactly what we expect - quickly go beyond 25Khz.

Please try out this new version, and if it seems to fix this problem, then go ahead and close the issue.

oskay, 2017-07-07:
O wow-- sweet! I'll try this out. :)

oskay, 2017-07-07:
**whoa**

EmbeddedMan, 2017-07-07:
Like, good whoa, or bad whoa?

oskay, 2017-07-07:
Lots of things that were not working are now working. Say... have you tried out the AxiDraw yet?

EmbeddedMan, 2017-07-07:
That's really, really great news!

I am planning on setting the AxiDraw up at work, and doing a demo for all
of my engineer coworkers. It's still in the box, waiting for some of the
fires to be put out. I'm planning to bring it out and show it off next week.

oskay, 2017-07-07:
Please contact me by e-mail when you're getting set up-- I'll want to set you up with the latest version. We have a good cross platform CLI interface working now too.

closed 2017-07-07 by oskay

EmbeddedMan, 2017-07-07:
Wow! Very cool! I will. Thanks Windell.