evil-mad/EggBot #79: Continued issues with LM command

oskay, 2017-07-09:
Following up on #78 (but separating this into a new issue, since these things seem quite unrelated to the initial topic there), I have found some continued unexpected behavior. This may represent a queueing issue.* 

To provide a very minimal example of the issue, I'm sending commands in a shell script.

The following works as expected: A single motor moves in the expected direction. There are two segments at different rates, both in the same direction so that you can tell that they are different by the sound:
```
#!/bin/bash
echo -e "SM,600,2032,0" > /dev/cu.usbmodem1461
echo -e "SM,1000,2032,0" > /dev/cu.usbmodem1461
```

We can also reverse the movement:
```
echo -e "SM,600,-2032,0" > /dev/cu.usbmodem1461
echo -e "SM,1000,-2032,0" > /dev/cu.usbmodem1461
```

We can also run lines 1-4 from the same script, and it works as expected. There are two movements in the same direction (faster, then slower), and it then it goes back in the same direction (again faster, then slower). This works as expected:

```
echo -e "SM,600,2032,0" > /dev/cu.usbmodem1461
echo -e "SM,1000,2032,0" > /dev/cu.usbmodem1461
echo -e "SM,600,-2032,0" > /dev/cu.usbmodem1461
echo -e "SM,1000,-2032,0" > /dev/cu.usbmodem1461
```

Now, we can try the same thing, using what _should be_ identical commands with the LM command.

This set of commands echoes the first two with SM, and appears to work properly:
```
echo -e "LM,290912465,2032,0,0,0,0" > /dev/cu.usbmodem1461
echo -e "LM,174547479,2032,0,0,0,0" > /dev/cu.usbmodem1461
```

This set of commands echoes the second two with SM, and appears to work properly:

```
echo -e "LM,290912465,-2032,0,0,0,0" > /dev/cu.usbmodem1461
echo -e "LM,174547479,-2032,0,0,0,0" > /dev/cu.usbmodem1461
```

However if we combine them, it does not work as expected:

```
echo -e "LM,290912465,2032,0,0,0,0" > /dev/cu.usbmodem1461
echo -e "LM,174547479,2032,0,0,0,0" > /dev/cu.usbmodem1461
echo -e "LM,290912465,-2032,0,0,0,0" > /dev/cu.usbmodem1461
echo -e "LM,174547479,-2032,0,0,0,0" > /dev/cu.usbmodem1461
```

The above set appears to execute out of order-- I believe that it's line order 1,4,3,2. And, I see no reason that this code should execute any differently than the same set with SM.

*Bonus points to me for using a word with five vowels in a row.

EmbeddedMan, 2017-07-09:
I have reproduced this issue. I'll track down what's going on.

BTW, do not ever let me play scrabble against you.

EmbeddedMan, 2017-07-09:
New version 2.5.3 created which fixes this problem.

The problem was that the LM parse code was writing the new command's data into the data structure of the previously parsed move which was sitting in the FIFO. Now we parse into a local move structure, then copy that move structure into the FIFO only when there's room.

oskay, 2017-07-09:
Okay, big wow again. With this change, I'm now able to run complex test patterns and have the results look excellent. And, thank you for the quick follow-up.

I'm simplifying the version history of this command in our docs to "First added in v2.5.0 with partial functionality. This command should only be considered to be functional as of v2.5.3."

closed 2017-07-09 by oskay