evil-mad/EggBot #80: Mac build script out of date

gregory-whaley, 2017-07-15:
The source URLs have changed from gatech.edu to uwaterloo.ca

oskay, 2017-07-15:
This is about the TSP scripts located at: /other/TSP-stipple/tsp_art_tools/

oskay, 2017-07-15:
Current downloads are at: http://www.math.uwaterloo.ca/~bico/qsopt/downloads/downloads.htm

We don't have anyone that is developing these tools -- it looks straightforward to modify the scripts if all of the updated file locations can be identified -- but I don't know that we can test them.

Amongst the issues: There only seems to be one version available for Mac download (as opposed to the old 32/64 bit versions) and I do not have a 32-bit mac available for testing.

oskay, 2017-07-15:
I tried compiling and was unable to get it to work with the updated links. 
If someone knows the current location of the 64-bit MacOS files and is willing to update the script and test the file, I'd be happy to accept the update. 

If not, we will think about removing this set of apparently broken scripts from the EggBot repository.

gregory-whaley, 2017-07-20:
I was able to get the updated script to compile on 64bit MacOS.

The linkern executable runs without error.

oskay, 2017-07-20:
I was not able to get it to work.

gregory-whaley, 2017-07-21:
The attached script worked for me.

oskay, 2017-07-25:
Can you please say what you mean by "the attached script"?

gregory-whaley, 2017-07-28:
Here it is.
[build-concorde-osx-0_2.sh.zip](https://github.com/evil-mad/EggBot/files/1181785/build-concorde-osx-0_2.sh.zip)

oskay, 2017-07-28:
Thank you, that seems to have worked. I made one change from the script that you attached, removing a couple of double slashes (`//`), that seems to not matter one way or the other.

closed 2017-07-28 by oskay