evil-mad/EggBot #83: (EBB) Add ability to give an EBB a 'name'

EmbeddedMan, 2017-08-12:
It would be really slick if, on a given computer, you could have multiple EggBots, AxiDraws and WaterColorBots all plotting away from separate instances of Inkscape at the same time.

To support this, some way of choosing which EBB a given Inkscape connects to is needed. I have no idea how to do this in Inkscape, or if it's even possible. But no matter what, EBB will need a unique identifier.

So, this feature request is to add a new set of commands. One for writing down a new 'name', and one for reading back that name. This name string will of course need to be stored in Flash, as these parts have no true EEPROM, but that's not a big deal. (I think) 

Many bonus points if this name also shows up in the device manager. :-)

oskay, 2018-05-09:
I'd like to get this wrapped up soon if we can; AxiDraw multi-machine support is almost ready and this would help. So far as I can tell, this feature is working and usable.

We should likely add a note in the docs discouraging programmatic setting of the name tag (flash lifetime yada yada). Also, it might be good to use some example _without_ an apostrophe, since those don't translate as well in the com port listings.

EmbeddedMan, 2018-05-09:
So, you say the feature is working and usable - but we haven't added an EBB firmware command to support the naming yet, have we? I don't see it in the release notes. Can you clarify what's left to be done?

oskay, 2018-05-09:
Look at the changes in the EBB_naming branch that you created?

EmbeddedMan, 2018-05-09:
I think I'd forget the names of my kids if they didn't remind me. So, if
the only thing to do just merge the branch and build a new version?

oskay, 2018-05-09:
I think that the firmware is built (but please double check)-- we'd need to merge, make those little updates to the docs, and build the windows updater program. I've already built a Mac updater for 2.5.4 -- please let me know if there is any reason _not_ to make `EBF_v254.unified.hex` the new version.

(Also on the topic of updater programs: please update your the "Updating Your Firmware" section on your main EBB page, with a link to the much-easier-for-most-people version at: https://wiki.evilmadscientist.com/Updating_EBB_firmware )

EmbeddedMan, 2018-05-09:
> com/Updating_EBB_firmware )

Ahh! Yes, that's a very good idea. Will do.

EmbeddedMan, 2018-05-10:
The branch has been merged. The windows updater has been updated. The SchmalzHaus EBB documentation page has been updated. I see no reason why 2.5.4 should not be the new version.

oskay, 2018-05-10:
Fantastic. Thank you!

closed 2018-05-10 by oskay

oskay, 2018-05-15:
I'm not seeing the Windows app. I'd expect it to be called EBBUpgraderGUI_v254.exe?

EmbeddedMan, 2018-05-15:
Sorry - forgot to include the executable. It should be there now. And yes,  you got the name right.

oskay, 2018-07-05:
I've been trying to connect the dots on this, but haven't quite succeeded in making things work right on Windows. 

With 2.5.4, I can write and read the device name successfully (Mac & Win), using the EBB commands. However, I'm not having the same luck with the 

Listing the port information via [pyserial](https://pythonhosted.org/pyserial/tools.html#serial.tools.list_ports.comports) on Mac, I get the following: 
port[0]: `/dev/cu.usbmodem1441`
port[1]: ` EiBotBoard_East`
port[2]:  `USB VID:PID=04D8:FD92 LOCATION=20-4`

This is *correct* as far as I can tell, and lets me successfully open a serial connection to that EBB alone.

Trying the same thing on Windows 10, I get the following:

p0: `COM6`
p1: `USB Serial (UBW-based) communications port (COM6)`
p2: `USB VID:PID=04D8:FD92 SNR=5`

Further, I don't see any mention of the name ("East") in the device manager. It seems like this is not quite working right.

EmbeddedMan, 2018-07-06:
It appears each os does this stuff a little differently. Actually, from what I can tell, all other OSes do it one way, and Windows does it a different way.

If you set the name of your EBB to "Brian123" for example, and you want to see this show up in the Device Manager, you need to right click on the COM port in DM, select Properties, go to the Details tab and pull down the "Bus reported device description" property. Then you'll see "EiBotBoard,Brian123".

It's definitely there under Windows 10. The tough part is figuring out how to access it from a given language. 

What gets shown on the actual Device Manager window as the name of the COM port is called the "Display Name" and it's set via the INF file, not via device descriptors in the device itself as far as I can tell. (There may be magic here that I just don't understand that would allow us to get a unique name in the Display Name field.)

I don't know any python, or how easy it is to access Windows C calls from within it, but this page does give some code snippets that look like they may provide this Bus reported device description value that we're looking for:

https://social.msdn.microsoft.com/Forums/lync/en-US/35ad96d5-df46-469d-be06-5ab78f090b31/how-to-get-the-bus-reported-device-description-of-a-device?forum=csharpgeneral

I really wish this were consistently implemented across OSes but as far as I can tell it isn't.

EmbeddedMan, 2018-07-06:
In addition, the standard tool for Windows to see what a USB device makes available to the OS is USBView (which can be downloaded from here : http://www.ftdichip.com/Support/Utilities.htm#MicrosoftUSBView)

The device descriptor for an EBB will list the full name "EiBotBoard,Brian123" as a string descriptor under the iProduct field in that tool.

EmbeddedMan, 2018-07-06:
After more research, it looks like WMI does not give this information. Instead, one has to use this call:
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/devpkey-device-busreporteddevicedesc
from the Setup API in Windows to get it.

oskay, 2018-07-06:
Interesting. I can indeed see it as "Bus reported device description" in the device manager. That's something to go on.

I also note that it lists "SNR=5" -- do you have access to update the serial number?

EmbeddedMan, 2018-07-06:
YES! I think the serial number is one of the other things that's not hard
to change on the fly. However, if you change the serial number, Windows
will assign it whatever the next unused COM port number is. So if you
change serial numbers a lot (to new unique values) you'l use up a lot of
COM ports, which may be just fine.

oskay, 2018-07-06:
So... what if we make the name tag also show up as the SN?

EmbeddedMan, 2018-07-06:
Right - I'll give that a go.

oskay, 2018-07-06:
I'll keep digging in the other direction as well. :)

EmbeddedMan, 2018-07-06:
This tool https://www.uwe-sieber.de/usbtreeview_e.html#download is much better than the simple USB View. It shows way more information about exactly what the OS knows about a USB device.

Interestingly, I see that when an EBB is 'fresh' (i.e. just programmed, never been given a name before) USB Device Tree Viewer shows the "Product string" as "EiBotBoard,****************" where the * are actually little circles. Like it's getting a string with the full 27 characters but the last 16 are invalid unicode or something? I'll need to look into that.

EmbeddedMan, 2018-07-06:
OK, version 2.5.5 is ready for your testing. The 'name' written using the ST command will now be used for the USB Serial Number field, as well as appended to the end of the device name field.

I also fixed the bug that was adding extra characters at the end of the device name field.

Give it a good test to see if you're able to query the serial number somehow from python or not.

If it doesn't work, I think we should revert the change, because every time you change the serial number you will get a new com port. (I'm up to about COM48 on my PC because of all of the testing today.) That may not be desirable behavior if the serial number field doesn't buy us anything.

oskay, 2018-07-06:
cool! Testing.

USB data on Mac:
port[0]: `/dev/cu.usbmodemNort1`
port[1]: `EiBotBoard_North`
port[2]: `USB VID:PID=04D8:FD92 SER=North LOCATION=20-6`

Interesting (1): The USB enumeration has changed.
Interesting (2): It's now reporting the serial number in the port[2] field.

On Windows:
port[0]: `COM3`
port[0]: `USB Serial (UBW-based) communications port (COM3)`
port[0]: `USB VID:PID=04D8:FD92 SER=NORTH LOCATION=1-9`

So-- it's working. And, I'd say _much better_ than it was before, since both platforms are on more equal footing. Looks like a winner.

I should be able to have a version of AxiDraw Control for you to try that can individually address different machines very soon, with this addition. Would you be interested to try it?

EmbeddedMan, 2018-07-06:
I would love to check it out, but I'll be honest, I likely won't have time
to actually set it up for a couple of weeks, so don't rush.

oskay, 2018-07-06:
Would you have time to roll the Windows updater for 2.5.5?

EmbeddedMan, 2018-07-06:
Yup, for sure

EmbeddedMan, 2018-07-07:
OK, should be good to go now. Try it out.

oskay, 2018-07-07:
Not working, thus far. Fails with runtime error: Dimension 1 should be between 1 and 0. Current value 1.

EmbeddedMan, 2018-07-07:
Hmm. I'm unable to reproduce that error. I just updated every EBB I own and
they all went fine without any problems.

Does it give any more information about the runtime error? Can you take a
screenshot of it?

oskay, 2018-07-07:
Well... after getting that error several times, it appears to be working now. :(
I guess we're good. :?