evil-mad/EggBot #86: Unknown tags in SVG file can cause error

gerv, 2017-09-10:
STR: try and print this file: https://cdn.thingiverse.com/assets/f0/cf/56/b7/ba/world_map_solid_continents.svg .
It prints if you copy and paste all the lines into a new blank Eggbot template.

The error is the following stack trace:

> Traceback (most recent call last):
  File "eggbot.py", line 1168, in <module>
    e.affect()
  File "/usr/share/inkscape/extensions/inkex.py", line 268, in affect
    self.effect()
  File "eggbot.py", line 200, in effect
    self.plotToEggBot()
  File "eggbot.py", line 449, in plotToEggBot
    self.recursivelyTraverseSvg( self.svg, self.svgTransform )
  File "eggbot.py", line 880, in recursivelyTraverseSvg
    self.sCurrentLayerName + '" unable to draw <' + str( t[-1] ) +
AttributeError: EggBot instance has no attribute 'sCurrentLayerName'

which is generated when attempting to execute this line of code:
> inkex.errormsg( gettext.gettext( 'Warning: in layer "' + 
	self.sCurrentLayerName + '" unable to draw <' + str( t[-1] ) +
	'> object, please convert it to a path first.' ) )

which appears to be an error which is thrown when encountering a node with an unknown tag name.

The problem appears to be in eggbot.py. Around line 505, self.sCurrentLayerName gets set if an "if" test is fulfilled, which seems to be trying to get the Inkscape layer name. It then goes on to unconditionally call self.recursivelyTraverseSvg(), which unconditionally references self.sCurrentLayerName in various error messages. However, it's at least possible that this value is unset, and so the above error occurs. The error which comes is one for an unknown tag or node, or some similar thing, so perhaps the file was not produced by Inkscape, or perhaps it was post-processed by some other processor which added the unknown node, or removed the Inkscape layer name information. (Although when you load the file, Inkscape has no problem saying it has a single layer.)

Either self.sCurrentLayerName needs to always have a value (probably better), or we need to test for non-nullness every time we try and use it.

gerv, 2017-09-10:
Apparently, the dodgy object concerned is a &lt;script&gt; object.

oskay, 2017-09-10:
Please state which version of the EggBot extensions you are using.

Also, I don't see a script object in the SVG file, if I right-click download the link above.

gerv, 2017-09-10:
EggBot extensions 2.8.0.

oskay, 2017-09-10:
And if you open the file with a text editor, you see a script there?

oskay, 2017-09-10:
I am not, thus far, able to reproduce the issue.

gerv, 2017-09-13:
I've found the problem:
<script xmlns="" id="custom-useragent-string"/>
It's a Firefox extension messing with my files :-(

Nevertheless, I argue that setting a value conditional on an if, and then later assuming that value will always be set, is a bug and should be fixed.

oskay, 2017-09-14:
Yes, it's important to right-click download SVG files, rather than to let them load in a browser before saving. 

If you can zip the file and post it here, then I'd hopefully have a way to reproduce the error locally.

gerv, 2017-09-14:
Github does not allow the attaching of SVG files, so I've renamed it to a TXT.
[world_map_solid_continents.txt](https://github.com/evil-mad/EggBot/files/1302231/world_map_solid_continents.txt)

oskay, 2017-09-14:
Thank you. (zip files are allowed, and tend to preserve weird encoding issues, which is why I suggested that approach.)

Looking at the file, I can see why we're not catching this already. The tag is not contained within the usual SVG tree that we're parsing. Instead, it's (hand-waving) in the header of the file, before certain variables to handle it are already set up.

This is apparently a very rare case, and should easily overcome by either downloading the file properly or by importing it into a fresh template. None the less, it would be good to handle it gracefully.

oskay, 2019-06-19:
Since I haven't come across another case like this in the (rather long) period since we looked at this, I'm going to close this issue. 

If anyone is affected by this, please feel free to leave a comment, and we may revisit it in the future.

closed 2019-06-19 by oskay