evil-mad/EggBot #90: Feature Request: Hatch fill individual SVG groups

DBraun, 2017-11-14:
My SVG is a bunch of sibling tags. Is it possible to use the hatch fill extension to fill each polygon individually but not make connections between polygons? I think I'm just using the settings incorrectly.
hatch_fill_svg.zip
[hatch_fill_svg.zip](https://github.com/evil-mad/EggBot/files/1469451/hatch_fill_svg.zip)

oskay, 2017-11-14:
Hatch Fill does not work well on sets of vanishingly thin objects. This is a known issue, and one that could definitely use improvement.

I was able to get your file to work with the following settings:

Spacing 2.0
Angle: 45
Connect: Yes
Range: 3.0
Inset: Yes
Inset by 0.5
Tolerance: 1.0

I was also able to get the individual slices to work (on a wider variety of settings) by ungrouping and selecting them individually. 

Also: Do we really have to do this twice?

DBraun, 2017-11-14:
Those settings are a big help thanks. My idea of using -45 as the angle was definitely bad. It's better to use 45 in this case and optionally also use a smaller hatch spacing. Sorry for the earlier cross post on evil-ad/axidraw. I realized after the fact that the hatch fill code was in this repo (and also that it's not so easy to delete github issues)

oskay, 2017-11-14:
It looks like better behavior for the extension would be to apply the hatch fill to each item within the group independently. I'm going to re-open this to remind us to work on that. :)

oskay, 2018-01-07:
Fix written and committed; if you download the current snapshot (click "clone or download" from https://github.com/evil-mad/EggBot ), and copy the latest version of eggbot_hatch.inx and eggbot_hatch.py into your extensions directory, it should now work properly.

closed 2018-01-07 by oskay