evil-mad/EggBot #94: Eggbot custom resolution

Polkic, 2018-02-22:
I was told to post my question here. I have a case where I am going to use custom gears. Lets say egg stepper will have ratio 2:1. One turn will need 6400 step. Where in inkscape extension can I add the multiplier x2. Thank you for help.

oskay, 2018-02-22:
You can set the resolution in `eggbot-conf.py`. Different resolutions for the two motors are not supported.

closed 2018-02-22 by oskay

EmbeddedMan, 2018-02-22:
I don't know the internals of the extension well at all, so I'm sure others
will have better suggestions.

But looking through the code a bit, it looks like

https://github.com/evil-mad/EggBot/blob/master/inkscape_driver/eggbot.py#L963

is where the same scaling factor is applied to both x and y. I think you'll
want to try modifying that (and the section immediately below it) to add an
extra x2 factor for one of the two axis (whichever has your 2:1 ratio) and
see if that works.

*Brian