evil-mad/EggBot #95: Error if " Start with pen centered" unchecked

Andy1978, 2018-03-09:
Release 2.8.1, dated 2017-06-07

If I run the extension with "Options" - "Start with pen centered" unchecked, I get a error message:

```
Traceback (most recent call last):
  File "eggbot.py", line 1169, in <module>
    e.affect()
  File "/usr/share/inkscape/extensions/inkex.py", line 283, in affect
    self.effect()
  File "eggbot.py", line 200, in effect
    self.plotToEggBot()
  File "eggbot.py", line 447, in plotToEggBot
    self.recursivelyTraverseSvg( self.svg, self.svgTransform )
  File "eggbot.py", line 509, in recursivelyTraverseSvg
    self.recursivelyTraverseSvg( node, matNew, parent_visibility=v )
  File "eggbot.py", line 565, in recursivelyTraverseSvg
    self.plotPath( node, matNew )
  File "eggbot.py", line 979, in plotPath
    if (plot_utils.distance(self.fX - self.fPrevX,self.fY - self.fPrevY) > eggbot_conf.MIN_GAP):
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'
```

I've added some debugging outputs (I never used python before so this might be noobish) and get
```
self.fX             = 164.59909
self.fPrevX         = None
self.fY             = 499.84871
self.fPrevY         = None
eggbot_conf.MIN_GAP = 1.0
```

So I guess there should be some code to initialise self.fPrevX and self.fPrevX around line 968, something like
```
if self.options.startCentered:
	self.fPrevX = self.svgWidth / ( self.step_scaling_factor )
	self.fPrevY = self.svgHeight / ( self.step_scaling_factor )

	self.ptFirst = ( self.fPrevX, self.fPrevY )
else:
	self.fPrevX = 0
	self.fPrevY = 0
	self.ptFirst = ( self.fX, self.fY )
```

oskay, 2018-03-09:
Confirmed. That's a bug. Part of the reason that we hadn't noticed this sooner is that (so far as we know) essentially everyone leaves this enabled, all the time. (Same goes for "return home when done.") 

The best way to fix it is a little less obvious. An easy fix is to fix both of those options to be true, and remove them from the control panel there. 

Question (for anyone watching this repository): Is there any compelling reason to _not_ remove both of those options?

Andy1978, 2018-03-14:
I'm absolutely fine to remove these two options. Apparently they never worked or broke long time ago without anyone noticing.

Every change breaks someones workflow ;-) https://xkcd.com/1172/

closed 2018-04-04 by oskay