evil-mad/EggBot #101: L3: 3-D (axes+pen servo) FIFO-enqueued synchronized low-level motion command

lsemprini, 2018-06-04:
Hi,

Thank you for your awesome EBB.

We would love to use our AxiDraw for 3-D drawing where the vertical (pen) axis is as accurate as possible given the capabilities of the servo mechanism.

We have many applications for this in mind, but one of them is simply to control a soft brush pen with varying vertical positions to give expressive brush thickness effects that vary during strokes---a particularly good application because vertical resolution is not critical.

Currently it appears that the firmware only has the concept of 0/1 up/down with a programmable servo position given to the SC command, and indeed we can program using SC to achieve a variety of positions...BUT...the SC command is documented as having immediate effect.  So unless I'm missing something, I don't see a way to synchronize pen servo changes with a/b or x/y changes.

Is there a way to do it now?

If not, I'd like to propose an L3 command that extends LM.

I am proposing to extend LM rather than XM/SM because it will be more powerful for users (albeit more complex) and I suspect it will be LESS work for EmbeddedMan :)

The details no doubt will depend on how the servo control works but here is a rough idea:

L3 – 3-D (axes + pen servo) Low-level Move

- **Command:** RateTerm1,AxisSteps1,DeltaR1,RateTerm2,AxisSteps2,DeltaR2,RateTerm3,AxisSteps3,DeltaR3<CR>
- **Response:** OK<NL><CR>
- **Firmware versions:** XXX and newer
- **Execution:** Added to FIFO motion queue
- **Arguments:**   (expand to explain how the servo can be controlled: units, step size, etc.)
- **Description:**   (expand to explain how the servo can be controlled: units, step size, etc.)

oskay, 2018-06-04:
Might be simpler to add an alternate form of SC that is added to the motion queue.

oskay, 2018-06-04:
Also, here are some known approaches to doing 3D moves with the existing firmware:
* Use shorter movements. The FIFO is only one command deep... so if you care to treat it that way, it can function as a system where your 3D move consists of many short moves, each with a different Z value. (This is much like how movements in XY are composed of many linear segments.) If your `SM` moves are much shorter in duration than the servo movement time (say under 20 ms), you can simply alternate updating the servo position and sending new movements. You can use the `QM` command if necessary to verify when the FIFO queue is empty in order to ensure that things remain synchronized. 
* You can send a pen lift (or lower) command directly before an XY move command. For example, the command `SP,1\rSM,1000,0,3200` will begin the Z move and then (almost immediately) begin the XY move, such that the two happen simultaneously in practice. Note that this use of the SP command does not specify a delay time. If you do specify a delay time (as in `SP,1,1000\rSM,1000,0,3200`), then the XY move will not begin until the Z move completes. The rate of the pen lift can be specified with the SC command.
* You can directly change the definition of (for example) the pen-down Z position, by giving a command like `SC,4,10000\rSP,1` that both configures the pen down position and moves to that new position. Note that the value of Z position that you send (10000 here) should be kept in the range specified in axidraw_conf.py, 7500 to 28000, in order to prevent damage to the mechanism. This type of command also be executed before an XY move, to give the same type of simultaneous motion.

lsemprini, 2018-06-04:
Hi oskay!  Thanks for your response.  

A queued version of SC would also be OK (the only disadvantage being no support for an acceleration parameter, but not sure that would even be physically possible to implement with the pen servo).

As to your known approaches (thanks for the list!), we had tried some of those but kept running into different issues:

- the strategy of sending a large number of SC+SP+tiny move commands, and waiting for a response before proceeding so that all 3 commands are basically treated as unqueued/immediate, is the first thing we tried, but because there is so much latency getting from our python software process through the OS USB layers, through the hardware USB, and getting the response back again, this led to pretty crazy bad performance.  Because the hardware has no motion queued up ever when using this strategy, it spends most of its time (most of its duty cycle) not moving, waiting for another motion command.  That unqueued approach also lead to such jerky movement in all axes on the actual device that the picture drawn by the pen was affected and notably not smooth (maybe due to extra vibration?).  It would be way, way better if we could somehow queue up movements in X, Y, and Z well ahead of time to keep the device happy and not have to worry about the USB latency.  Note that in this case it is high latency (as opposed to unpredictable latency) that is the problem; contrast that with the last strategy described below.

- the strategy of firing off a long-duration set of XY move commands at about the same time as a single long pen-up/pen-down command works if we want to move the pen linearly during that period of time,  but of course that depends on the content being rendered.  If we want the motion in the Z direction to be "curved" (anything but a straight line over time), as we often see when stroking with a brush, it won't work.  But thanks for mentioning this strategy.

- another strategy we thought of was whether we could issue the usual high-performance, very-very-queued up XY motion commands ahead of time and then try to issue the immediate pen commands at the "right" time.  This has two big problems.  One is that no modern OS allows our software process reliable scheduling so that we can run the code to send the pen commands at the right time.  Another is that even if we had real-time scheduling in the OS, in order to know when to send the pen commands we would have to know the exact value of the latency through the whole system (OS USB software/hardware, device USB software/hardware).  But that latency is unknown (to clarify, for this strategy, the latency could be very large with no problem, but the fact that its value is unknown is a big problem).  One could possibly rig up a closed-loop test using the QM command to try to measure the latency at a given moment, but I'm sure sometimes the latency changes due to different congestion conditions at different points in the pipeline.  It seems basically hopeless to get this approach working, even if we had proper OS scheduling, but since we don't even have real-time scheduling, it's moot.

So thanks again for sending those "known approaches."  I appreciate the time you're taking to help us and others, and when I wrote this comment I just wanted to write down the issues with those approaches to help motivate why doing a queued pen-movement command would be worthwhile.

Thanks!

EmbeddedMan, 2018-06-04:
Thinking about this some more, I like the L3 approach the best. I don't
think it would be that much more work (saying that without the code in
front of me), and it would give the best results.

I'll find some time to play around with it and see what can be
accomplished. Honestly, all of the functionality is actually in place
(moving the steppers some amount per unit time, and moving the servo some
amount per unit time), you just can't do both once with the current code.

*Brian

oskay, 2018-06-04:
If we do, it would be nice to add matching `S3` and `X3` commands as well.

EmbeddedMan, 2018-06-04:
Completely agreed. That shouldn't be too hard either.

*Brian

EmbeddedMan, 2018-06-05:
OK, after looking at the code for a few minutes, this isn't going to be
trivial. The code that handles RC servo moves currently is broken out into
two sections. One runs at the high ISR rate (25KHz) and pulls RC servo
moves off of the motion queue. When the current command is an RC servo
move, then it sets a bunch of values which start that RC servo moving and
then it's done.

The second piece of code runs at the low ISR rate (1KHz) and it looks at
the values set up by the first piece of code. Each time the particular RC
servo channel's slot comes around, it sees if it needs to change the
position, and if so by how much.

There's nothing wrong with this, but in order for the L3 command to work,
we're going to have to add a flag into the second piece of code. If the
currently executing command is an L3 command, then we can't pull the next
command off the motion queue until both steppers are done moving AND the
servo is done moving (this is the point of the new flag, which will be set
by the low ISR when a servo move is complete).

I need to be careful not to change the behavior of the existing SP commands
at all, so when I get this code done I'll want you guys to test it out in
multiple real world situations before we call it good. But I think this
should not be too hard - we'll create a new kind of command that does all
3, and we will expand the structure put onto the motion queue to include RC
servo information.

One thing we have to figure out is the RC servo channel. Remember, we can
have up to 8 RC servos going at the same time on the EBB. One of them is
normally used for the pen up/down servo. Should the L3 command have an RC
servo channel parameter that lets you choose any of the 8 channels to apply
the RC servo move to? Or should we just hard-code the L3 command to always
use the RC servo channel that the pen up/down (SP) is tied to?

We could (easily) of course make this take a set of 3 parameters for all 8
of the RC servo channels, so you could have all 8 doing different things at
different times, speeding up, slowing down, etc. That would in essence give
you a 10 channel motion control system, where 2 of them are high resolution
full rotation, and 8 of them are low resolution partial rotation. Hmmm.
Other than like weird puppetry, I'm not sure I can think of a good use for
that. Since we're not really trying to make the EBB the world's best RC
servo controller, maybe just limiting the command to one RC servo channel
is good enough. But I like the idea of allowing it to act on ANY of the 8
channels.

*Brian

oskay, 2018-06-05:
What about the idea of making a version of SC that goes into the FIFO?  
<code>SC, <i>value1</i>, <i>value2</i>[, <i>addtoFIFO</i>]&lt;CR&gt;</code>

EmbeddedMan, 2018-06-05:
> SC, *value1*, *value2*[, *addtoFIFO*]<CR>

That would be a much bigger change. Currently none of the various SC
commands enter the FIFO, and there is nothing on the other end to 'execute'
them. Right now, all SC commands are simply executed as soon as they are
parsed.

If I were to do this whole thing over, I'd make all of the commands (or at
least most of them) be able to be executed immediately, or go into the
motion control queue and executed at their proper point in time. Doing that
now would, I fear, result in a huge refactoring of some parts of the
codebase, but would make this thing far more flexible. Ahh, hindsight.

*Brian

oskay, 2018-06-05:
One more alternative approach, what about adding the third axis directly to the SM command? (`S3`?) 

It could be something like:
<code>S3,<i>duration</i>,<i>AxisSteps1</i>[,<i>AxisSteps2</i>][,<i>servoFinal</i>]</code>

The advantage of this would be that it is *time definite*, so when it comes time to execute that command from the FIFO, it could be modeled as doing the following existing things:

(1) Execute an immediate change of the pen-up or pen-down position (whichever we're not in), the same way that we do with `SC,4`/`SC,5`
(2) Execute an immediate change of the pen slew rate, the same way that we do with `SC,10`. This should be straightforward, since we *know* both the total time duration in ms and the difference between the pen-up and pen-down positions.
(3) Execute an immediate start of the pen vertical motion, the same way that we do with `TP`.
(4) Execute the remainder of the SM command.

EmbeddedMan, 2018-06-05:
> S3,*duration*,*AxisSteps1*[,*AxisSteps2*][,*servoFinal*]

Yes, this is in essense how the S3 command will need to work. But
internally it will compute all of the parameters to an L3 command, and then
call L3 (just like SM currently computes all of the parameters for the LM
and then calls LM).

*Brian

oskay, 2018-06-05:
I'd err on the side of keeping it as simple as possible. Making an S3 variant that supports a single third axis will benefit more users than the L3, most likely.

EmbeddedMan, 2018-06-05:
Absolutely. The simpler the API, the simpler the code and the testing.

lsemprini, 2018-06-06:
Any APIs above fine with us!

By the way, any ideas where to look to learn the model number and/or capabilities of the particular servo used for the AxiDraw pen (vertical resolution, accuracy, etc.)?  Assuming the EBB outputs the correct pulse width every 24ms, it will be interesting to compare the actual performance against what is spec'ed, but I'm not sure where to begin in looking for the spec.  Couldn't find servo specs on AxiDraw's website.

oskay, 2018-06-06:
Standard AxiDraw pen lift servo is a Tower Pro 9g servo motor.

lsemprini, 2018-06-07:
Cool thanks.  Looks like even Tower Pro doesn't document their accuracy/precision/etc.  Chalk that up to the $4 price I guess.  Have you ever heard of anyone modding their AxiDraw with a different, more expensive servo with more accuracy?

For others wanting to do 3-D work, here's what I found on the TowerPro SG90 9g servo's behavior:

https://github.com/tardate/LittleArduinoProjects/tree/master/playground/ServoTest

https://forum.arduino.cc/index.php?topic=353826.0

https://www.quora.com/Is-there-any-better-alternative-to-TowerPro-SG90-servo-motors

https://servodatabase.com/servo/towerpro/sg90

http://akizukidenshi.com/download/ds/towerpro/SG90.pdf

oskay, 2018-06-07:
SG90s are basic, but widely used for reasons including the low cost.

They are well modeled by assuming that they have 10-20 available positions that they can move smoothly between. We have tried a lot of other servos, including some upwards of $25 each (wholesale cost), but you really have to go _much_ higher in cost before getting to a significant performance advantage. Higher end servos tend to get heavier and slower (metal gears, etc), then to digital, well before getting to faster motors, then digital then, finally, to encoders which are what would make the biggest difference.

lsemprini, 2018-06-10:
Also, for others who are waiting for the new command that moves all 3 axes at once with perfect sync, here are some pointers about how to sort of do it (with big limitations noted above) using existing commands:

I tried various orders of commands but the only order that actually made the AxiDraw move in XY and Z simultaneously for me was:

SC,10,servo_rate
SC,4,pen_down_position
SP,1,0   # pen down: third arg says duration is 0ms
XM,milliseconds_to_do_xy_move,$x_change,$y_change
... immediately transmit N more copies of this sequence for the next moves ...

I found that if I attempted to issue the XM first, the SP always waited for the XM to finish, even if XM duration was 0.  Secondly, I found that I had to issue the SP with a delay of 0, otherwise the XM would wait for the SP to finish.   

Finally, and most confusingly, somehow magically if I issue the above series of commands in order repeatedly, the device does moves in the right sequence (that is, the next Z+XY move does not begin until the previous one has finished) even though that would seem to contradict the documentation, which states that XM is queued, so in theory the SC,SC,SP for the next move should take effect immediately, ruining the previous move.  But somehow it works.  Not sure why.

This will all become so much cleaner with a queued 3-D command!

lsemprini, 2018-07-09:
Howdy.  We created a website showing videos and images of some of our early results with the existing 3-D support to pique people's interest about what is possible with full 3-D support, provide sample code, and get people thinking about other possible applications for 3-D AxiDraw.  Enjoy!

https://lurkertech.com/3daxi/

lsemprini, 2024-05-27:
Hi, great to see all the work over the last years including some other changes ( #64  #120  ) partially in anticipation of 3-D function.  Still very much interested in finding a way to do smooth 3-D motions.  Just curious how close we are to that nowadays?  Anything we can do to help?

EmbeddedMan, 2024-05-27:
The 3D_Moves branch has the very beginnings of 3D support: basically just the framework for three new commands (S3, X3 and L3). No actual functionality there yet. But that was last touched in 2017. 3D moves isn't at the top of my priority list at present unfortunately. I wish that adding in a third axes were simple and straightforward, but it's not. There are also competing needs here : the commercial machines sold by Bantam Tools (and existing AxiDraw users) typically desire the 'fastest' moves possible - meaning we spent an enormous amount of time optimizing every code path (at the assembly level) that affects how quickly we can reliably execute stepper move commands. Adding in a third axes (even though it's servo based and not stepper based) will add logic to the time critical ISR that decides when a step is taken, thus possibly slowing down non-3D moves. This effect would need to be minimized of course, but that increases complexity. The other factor I can see right now that complicates things is that with the v3.0.x EBB firmware, we highly optimized RAM usage so that we could expand the motion FIFO as large as possible (up to 32 commands now). Each of those spots in the FIFO is a fixed size structure based on all necessary parameters for 2 axis of motion. The servo moves are not tracked alongside the stepper moves currently (which is why we don't have 3D moves already), which means that adding them increases the command structure size, thus decreasing the possible FIFO elements we can store in RAM. 

None of this is insurmountable, but based on the last 2 years of development from 2.81 to 3.0.3, I can tell you that it is surprising how complex these changes quickly become. 

So bottom line, I don't have a great answer for you: we are not close to having 3D moves working at this point, and I'm not sure when it will become the top priority for my development efforts.