evil-mad/EggBot #102: Hershey bug

gisep, 2018-07-05:
Hi,
i'm finding problems in tracing a gcode text path with spaces using the "Hershey Text" extension to do the text and "J Tech Photonics Laser Tool" to trace it.

The same problem is described here:
https://bugs.launchpad.net/inkscape/+bug/1646595
where Mitch Bradley suggests this patch:
https://launchpadlibrarian.net/295778686/hershey.patch

But i don't know how to run it on Windows 10.

Can you help me?

thanks,
giuseppe

oskay, 2018-07-05:
Use the current version.

closed 2018-07-05 by oskay

gisep, 2018-07-10:
Thank you buddy, i will try it soon :)😊

A: evil-mad/EggBot
Cc: gisep; Author

oskay, 2018-07-10:
The patch that you were describing was incorporated into the EggBot code here (but not yet in Inkscape) some time ago, so if you download the repository  and manually install the current versions of hershey.py, hershey.inx and hersheydata.py within your copy of Inkscape, you should be good to go.