evil-mad/EggBot #103: Add RC servo power control functionality to EBB firmware

EmbeddedMan, 2018-07-12:
New versions of the EBB will have the ability to enable/disable power to the RC servo connector (RB1) by software control. Pin RA3 on the MCU will be used to control a high side FET to the 5V servo power.

The desired behavior for this new feature is as follows:

(1) Power to the servo will be off at reset (like the steppers).
(2) Power to the servo will be switched on whenever steppers motors are enabled, or whenever there is a command sent that changes the Z position.
(3) Power to the servo will be switched off when the "motors" are disabled.
(4) Power to the servo will be automatically switched off after some period of time elapsed since the last action moving it. A good default might be one hour.
(5) A command will be used to change the power-off time to other values. It could be set in minutes or hours (or whatever timer-expiry value is convenient), and have a "negative" value option to disable auto power-off. It would not be necessary to save this value beyond reset.

EmbeddedMan, 2018-07-12:
A proposed set of new commands:

QR - Query RC Servo power (returns a 1 or 0)
SR - Set RC Servo power timeout. Will take one parameter, an unsigned 32-bit int, in milliseconds. Sending 0 will disable power timeout.

Of course, PO,A,3,1 will always turn on the servo power, and PO,A,3,0 will always turn it off (i.e. direct control, if anybody ever needs that.)

EmbeddedMan, 2018-07-13:
EBB v2.5.6 firmware has been posted to GitHub. Please give it a try when you are able with the new EBB hardware. If it works like you want, let me know and I'll build a new Windows updater for it.

oskay, 2018-09-07:
Here are some detailed suggestions for how this the new version should work, in addition to the notes above. Please feel free to disagree, clarify, and suggest better approaches. :)

* Call this EBB firmware v 2.6.0 ? This is a major new feature, and saying that the new EBB version requires 2.6+ will be easier than saying v2.5.6+.

* Suggestion: Add a direct command to toggle the power state.
	- <code>SR, <i>timeout</i> [, <i>state</i>]&lt;CR&gt;</code>  
	- <code><i>state</i></code>: 0 or 1, to turn power off or on.
	- The 0 and 1 state codes perform the same action as `PO,A,3,0`, `PO,A,3,1`.

* For boards with servo power control:
	- Servo power is off at reset.
	- Default timeout period is 15 minutes (900 000 ms)
        - Servo position is at "pen up" state, should power be enabled.

* For boards without servo power control:
	- Servo power is on at reset (obviously).
	- Servo position at "pen up" state.

* A change to the default pen-up, pen down positions:
	- SC,4,22565 ("Pen up: 70%" in AxiDraw conventions)
	- SC,5,15302 ("Pen down: 30%" in AxiDraw conventions)
	- These values will work much better than existing presets on all of our EBB-using machines.

EmbeddedMan, 2018-09-07:
All seems very reasonable. I'll see if I can get these in this weekend.

oskay, 2018-09-07:
Fantastic; thanks!

EmbeddedMan, 2018-09-08:
One thing - the software can't tell what type of board it's running on. So we can't do anything 'different' between boards that have servo power and those that don't. I think this just means that no matter what type of board you're on, port A bit 3 (servo power) will be on at bootup (reset) and will shut off 15 minutes after the last servo move. Hope that's OK.

oskay, 2018-09-08:
Yes, of course. Hence the "obviously" above.

EmbeddedMan, 2018-09-08:
OK, cool. Just making sure that it was clear.

EmbeddedMan, 2018-09-08:
OK, new firmware image for v2.6.0 is in the repo. Please test out and make sure it work as you expect it to (it passed all of my bench testing here). If so, let me know, and I'll create a Windows updater for it.

oskay, 2018-09-09:
Great; We'll give it a workout today. Thanks for the quick work!

oskay, 2018-09-09:
It looks like the servo power is _on_ at reset.

EmbeddedMan, 2018-09-09:
OK, I take it this is not what you wanted. I will change it so that we will
default to the up position in terms of pulse width, but the without
actually doing a "move" there (which is what's kicking the power on right
now at boot).

*Brian

oskay, 2018-09-09:
Yes; from the list above: "Servo power is off at reset."

EmbeddedMan, 2018-09-09:
Done. Check it out and see if it works for you.

oskay, 2018-09-09:
Great-- testing now. (And, so far so good!)

oskay, 2018-09-09:
I haven't tested everything possible, but it certainly looks like the basic functions are all working properly. I think that we're good to go.

Since 2.5.6 wasn't distributed, I think that it's safe to update the docs to say that the new features are "added" in 2.6.0,  if that's acceptable to you.

EmbeddedMan, 2018-09-09:
Absolutely, that's just fine with me. You can edit those docs however you
think reads best.

I'll create a Windows installer for 2.6.0 shortly.

EmbeddedMan, 2018-09-09:
OK, windows updater is up.

oskay, 2018-10-26:
This function is updated and working well. The current firmware update tools have been updated as well.

closed 2018-10-26 by oskay