evil-mad/EggBot #113: Return home command?

oskay, 2019-01-09:
We now have the `QS` command, which might give you a response like "15000,13000". 
You can (with some exceptions) use that to return to the starting point with a command like "SM,5000,-15000,-13000".*

That creates the possibility to add an automatic command to return home command, which is something that has been requested on occasion.

That could be as simple as: <code>HM,<i>step_rate</i>&lt;CR&gt;</code>

We would want to specify the speed as a step rate (or frequency rather than as a total time duration, because the total time would depend on the initial position. 

Now the tricky part: those exceptions. If `QS` returns "22000,1" (for example), and you ask it to move home at a rate of 11000 steps/second, that would nominally take 2 seconds on motor 1, but gives an invalid step rate (0.5 steps per second) on motor 2. So this can't directly map to `SM` -- it isn't actually trivial.

A saving grace is that this is a return-home type move, and could be a  "[dog leg](https://www.cnccookbook.com/g00-g01-cnc-g-code/)" type move rather than a true interpolated (straight line) move. And, so long as we allow the <code><i>step_rate</i></code> parameter to be interpreted as a *maximum* step rate, we have some flexibility. 

One way to do a dogleg move like that would be to do two moves (in pseudocode):

If (|steps1| > |steps2|)
{ 
durationA =  |steps2| / rate
durationB =  (|steps1| - |steps2|) / rate
SM(durationA, sign(steps1) * |steps2|, steps2)
SM(durationB, sign(steps1) * (|steps1| - |steps2| ), 0)
}
(etc)

A "sneaky" way of doing this (though not sure if it's actually plausible) would be to have the command put the second SM into the FIFO when it executes, rather than leaving the FIFO free.

*This method does additionally require that the machine can't lose power, be moved by hand with the steppers off, or lose position control by other means such as running the physical limits of travel.

*Aside*: We might want to consider resetting (ala `CS`) the step count if EM is called to change the resolution or turn off the motors. That would prevent some kinds of disasters that this command might otherwise allow.

EmbeddedMan, 2019-01-10:
I like the idea, and I have some thoughts on various ways it could be implemented internally. I'll do some testing this weekend and see what works.

Also agreed about clearing the step counts on CS commands.

EmbeddedMan, 2019-01-13:
OK, a new version (v.2.6.2) is waiting for your review in Branch_Issue113. It implements the HM command (including the splitting of the operation into two parts when necessary). Please try it out and let me know if you have any issues with it.

EmbeddedMan, 2019-03-14:
Any time to test out this new command? PR #114 is ready to merge in if this works for you.

oskay, 2019-03-14:
In limited testing, this does appear to be working correctly.

Also, really awesome to have this. :)

EmbeddedMan, 2019-03-14:
Most excellent. Feel free to merge this PR then.

oskay, 2019-03-14:
Since this incorporates the QG change, I'd prefer to wait until the issue with that is resolved.

closed 2019-03-16 by oskay