evil-mad/EggBot #115: Mac: not able to connect to EggBot

NSchnitzler, 2019-03-07:
My Setup is:
Inkscape 0.92
EggBot Software: Version 2.8.0 (manual copied into the extension folder)
MacOS 10.14.3
Python 2.7.10
pyserial 3.4
used port /dev/tty.usbmodem1442301

The EggBot is reacting on manual send commands in a terminal program, but Inkscape always returns:
Failed to connect to EggBot. :(

Has anyone an idea how to solve this or how i can improve the debugging?
thanks in advance

oskay, 2019-03-07:
Try deleting the "serial" folder from the Inkscape Extensions directory.

NSchnitzler, 2019-03-07:
thanks for your answer, 
but it didn't work

oskay, 2019-03-07:
Can you please tell me what version firmware you have? Use the "V" command.

Can you please double check that the EggBot software within Inkscape shows v 2.8?

Communicating separately in the terminal *can* leave the port in a "busy" state, so make certain that you've fully closed the connection before trying again in Inkscape.

jhitesma, 2019-03-07:
I'm actually having a similar situation with a new eggbot I just built - I'm not using an official eggbot board though so was hesitant to comment...and I apologize now if this is considered off-topic or doesn't help - but in hopes that this could be the same issue and more info might help I'm going to go ahead and share.

I am running v2.8 of the extensions.  On a mac with Inkscape 0.91

I have an eggbot I built from scratch 3 years ago running a version of the "EggDuino" firmware which I modified to run on a RAMPS board:  https://github.com/jhitesma/EggDuino/tree/jth-ramps  It connects and works just fine.

I just built the "Egg painter mini" (https://www.thingiverse.com/thing:2245428) mostly because I had a bunch of spare 28byj steppers on hand and was looking for something to do with them.  This is currently running the original authors own modified version of eggduino and like the OP here it responds correctly to commands from a serial monitor - but gives the failed to connect message when I try to connect to it from inkscape.

I've done some poking in egg_serial.py and added some extra inkex.errormsg() to help me dubug what's going on.

I was able to confirm that modifying the VID/PID to match the arduino UNO gets it past findPort().  However when it gets to testPort the response to the 'v' command is always coming back blank....even though if I connect with the arduino serial monitor and send a v I get the expected response that's identical response my RAMPS based eggbot returns.

I added a few more inkex.errormsg()'s in testPort() to try and figure out what's going on...and what I found is that with the uno based build comPort is coming across as a string while with the RAMPS based build I'm getting an object.  But I can't figure out why that could be happening.  They're both being detected by VID/PID (i'm editing ebb_serial.py and restarting inkscape when switching between them) so it appears it's the same:

					EBBport = port[0] #Success; EBB found by VID/PID match.

For both boards...but on one it's returning a string and failing on the other it's returning an object and working.  

I'm not very familiar with python though so I'm running out of ideas on what to try next.  (other than firing up one of my windows machines and seeing if the same thing happens there.)

oskay, 2019-03-07:
Yes, none of the software here targets those boards.

NSchnitzler, 2019-03-09:
the bottom of the Readme:
https://github.com/jhitesma/EggDuino/tree/jth-ramps
solved my problem.
Thank you

oskay, 2019-03-09:
So this wasn't about Eggbot? Yikes.

closed 2019-03-09 by oskay