evil-mad/EggBot #117: Fatal Errot when trying to print single layer

schlion, 2019-05-05:
When trying to print a layer 1 from the template file, i get this error
`Traceback (most recent call last):
  File "eggbot.py", line 1154, in <module>
    e.affect()
  File "inkex.py", line 283, in affect
    self.effect()
  File "eggbot.py", line 220, in effect
    self.plotToEggBot()
  File "eggbot.py", line 440, in plotToEggBot
    self.recursivelyTraverseSvg( self.svg, self.svgTransform )
  File "eggbot.py", line 499, in recursivelyTraverseSvg
    self.DoWePlotLayer( node.get( inkex.addNS( 'label', 'inkscape' ) ) )
  File "eggbot.py", line 898, in DoWePlotLayer
    if str.isdigit( CurrentLayerName[:stringPos] ):
TypeError: descriptor 'isdigit' requires a 'str' object but received a 'unicode'
`
This seems to be related to #40.
I have added a possible fix as a pull request #116

oskay, 2019-05-05:
Not immediately able to reproduce the issue-- please follow up either here or in the PR discussion.

schlion, 2019-05-05:
I'm using Inkscape 0.92.4 (5da689c313, 2019-01-14) on a Windows 10 machine.
When trying to print layer 1 (as is default) from the EggBotTemplate.svg I get this error.
I had installed the Plugin Version 2.5 before and then installed 2.7.1 over it. Possibly residual files are causing probelms?

oskay, 2019-05-05:
Residual files are likely not an issue. Are you using 2.7.1 now?

oskay, 2019-05-05:
If so, definitely update to 2.8.0

schlion, 2019-05-05:
Ah I see. #40 was only fixed in 2.8.0
Using 2.8 fixes my issue. Thanks a lot!

closed 2019-05-05 by schlion