evil-mad/EggBot #123: Status of Inkscape 1.0 support?

jbaehr, 2019-10-08:
A few weeks ago Inkscape published its [first beta of the upcoming 1.0 release](https://inkscape.org/release/inkscape-1.0/?latest=1). Especially for Mac users this is a reason to celebrate as it requires no more X11, but comes as a (more or less) native app.

In the repo I've seen that @oskay already created an [`Inkscape_v1` branch](https://github.com/evil-mad/EggBot/tree/Inkscape_v1). What's the status here? Do you need a helping hand?

Is there any best practice in bringing the code from the git repo into Inkscape?
I've managed to get the extension show up by grapping the py files from this repo and the plotlink repo, the `svg_fonts` dir from the `EggBot_extensions_v281.zip`, them some patching to get past relative-import issues (basically going back to the plot_utils from the zip), fetching `axidraw_svg_reorder.py` from by Inkscape-0.92 (installed from your bundle), ... All this doesn't feel right, though, but I don't want to reinvent the wheel if you have some established workflow to share.

I've also noticed that Inkscape doesn't seem to find it's python executable (the above linked bata-1 .dmg has py3 embedded) when started from the terminal (I did that to see error messages written to stdout), but it works when launching the app-bundle through Finder.  That's not an EggBot issue but may be helpful to other "early adopters".

oskay, 2019-10-08:
I've updated the release with a snapshotted bundle for 1.0.

Everything seems to be working fine-- we've had an installer available upon request and had been hoping to get a little more real-world feedback (confirmation of working and/or error reports) before just posting it publicly. _Things that we don't know about may still not be working correctly._

closed 2019-10-08 by oskay

jbaehr, 2019-10-09:
Thanks for the update.

The updates to `eggbot_hatch.py` doesn't seem to have hit the repo yet, though (i.e. in git I still see the old version. Missed that file on commit?).

I've got troubles with both versions, the older from this repo's `Inkscape_v1` branch, as well as the newer one from the v2.8.1 release's `eggbot-inkv1.zip`. Before starting investigations I want to make sure to work on up to date files, though.

oskay, 2019-10-09:
Good catch. https://github.com/evil-mad/EggBot/commit/00dcedd7269f66e41da9ac9222f3374d70769faa

oskay, 2019-10-09:
As far as I can tell, hatch fill is working correctly.

jbaehr, 2019-10-10:
Hatch is not (directly) the one to blame. Rather the lxml shipped with your zip only contains linux binaries (e.g. `etree.cpython-37m-x86_64-linux-gnu.so`) but none for mac-os (e.g. `etree.cpython-37m-darwin.so`)

I tried installing lxml via pip in my system's python (I have manually a python-3.7 installed, for other reasons; Apple still ships 2.7). This gave me the macos binaries; I replaced `.../inkscape/extensions/axidraw_deps/lxml` with the version from my site-packages (`pip3 show lxml` will show the location).
Here is the zip-archive pip downloaded: https://files.pythonhosted.org/packages/bd/9f/6cda4672d3ad1aa4cf818ab8401a763787efba751c88aaf4b38fc8f923bb/lxml-4.4.1-cp37-cp37m-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl

This resulted in a new error:
```
The fantastic lxml wrapper for libxml2 is required by inkex.py and therefore this extension.Please download and install the latest version from http://cheeseshop.python.org/pypi/lxml/, or install it through your package manager by a command like: sudo apt-get install python-lxml

Technical details:
dlopen(/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/axidraw_deps/lxml/etree.cpython-37m-darwin.so, 2): no suitable image found.  Did find:
	/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/axidraw_deps/lxml/etree.cpython-37m-darwin.so: code signature in (/Applications/Inkscape.app/Contents/Resources/share/inkscape/extensions/axidraw_deps/lxml/etree.cpython-37m-darwin.so) not valid for use in process using Library Validation: mapped file has no cdhash, completely unsigned? Code has to be at least ad-hoc signed.
Traceback (most recent call last):
...
```
So, the Inkscape-1.0-beta1 MacOS binary is signed, the lxml binaries are not. At least, that how I read this.

Using python3 in the Terminal, a `from lxml import etree` succeeds, though. And now the interesting part: *Now it works in Inkscape, too.* I can't say, whether the use of the lib in python's REPL, or the simple rety did the trick; I cannot reproduce it. It continues to work even if I place the original linux binaries from your zip back in place; and I don't know which caches to clean (the `__pycache__` dirs in `...inkscape/extensions/**` contain only python byte code (`*.pyc`) and no lxml or etree)

Disclaimer: I'm on MacOS 14.9.6. With XQuartz and Inkscape 0.92.2 EggBot works fine. I do this tinkering out of curiosity and with the aim of getting rid of X11.

oskay, 2019-10-10:
This set of extensions _relies_ on the lxml in your Inkscape installation. You should have the same results if you remove the copy of lxml.