evil-mad/EggBot #124: Math error in SM command for edge case input parameters

EmbeddedMan, 2019-11-06:
If you use the command "SM,6000,131071,0" in EBB firmware version 2.6.3 or below, you will get:
Major malfunction Axis2 StepCounter too high : 4294967295
as an error.

EmbeddedMan, 2019-11-06:
It turns out this bug needs a certain set of conditions to be triggered:
1) The Axis2 step count parameter must be 0
2) The Axis1 step count parameter must be greater than or equal to 0x1FFFF (131071)
3) The combination of the Duration and Axis1 step count parameters must create a valid step speed (<=25K steps/s)

If all of those conditions are true, then the bug is triggered and what should have been valid input is rejected with the nonsense error message listed above.

oskay, 2019-11-06:
(1.) isn't required.

EmbeddedMan, 2019-11-06:
True - it's slightly more complicated than that though. For example, SM,6000,131072,1 does NOT produce the error, it produces a different (correct) error of "!0 Err: <axis2> step rate < 1.31Hz"

In any case, the exact parameters necessary to produce the error are probably secondary to getting it fixed properly. PR coming in a minute.

oskay, 2019-11-06:
That particular case (`SM,6000,131072,1`) is already handled by your part (3.) above, which looks correct to me.

EmbeddedMan, 2019-11-06:
Please try out the v2.6.4 firmware on the issue 124 branch. I'm able to test a few SM commands by hand here, and they appear to work, but it should get some time on an EBB or AxiDraw before the branch is merged to make sure I didn't mess something up.

oskay, 2019-11-06:
In some simple tests plotting some sample files, I don't see anything amiss.

oskay, 2019-11-06:
Thanks for the quick response on this!

closed 2019-11-06 by oskay