evil-mad/EggBot #126: Update servo timeout behavior

oskay, 2019-11-30:
Following up on #103, it looks like the servo timeout was not actually implemented according to the original plan there-- which was that the "Power to the servo will be automatically switched off after some period of time elapsed since the last action moving it."

What is actually implemented is that "Power to the servo is be automatically switched off after some period of time elapsed since the last action moving it or the stepper motors." 

This implementation does not allow the servo to be off while moving the steppers.

The [documentation](http://evil-mad.github.io/EggBot/ebb.html#SR) appears to correctly describe the present behavior

I suspect that there's no harm in making a change to the **SR** command such that it waits only until after stepper movements, since the computer can always turn it *on* for longer.

EmbeddedMan, 2019-11-30:
Whoops. OK, on it.

EmbeddedMan, 2019-11-30:
New version 2.6.5 created with this issue fixed. Windows updater created. Docs updated. Pushed. Please test and merge if it's working as intended now.

EmbeddedMan, 2020-01-20:
@oskay If you've had a chance to test this new version, please merge the Branch_Issue126 in.

oskay, 2020-01-20:
https://github.com/evil-mad/EggBot/pull/129

closed 2020-01-20 by oskay