evil-mad/EggBot #131: Configure Pen Servo

aarondfrancis, 2020-02-24:
Hey y'all,

I'm trying to do something a little unorthodox (I think?) with the EBB board, and that is use one of the other servo outputs as the pen servo from time to time.

Based on what I can tell from the docs, it looks like you can use any of the port B pins when using the [SP](http://evil-mad.github.io/EggBot/ebb.html#SP) command.

But QP and TP don't offer that same functionality.

I'm trying to get to the point where I can control multiple pens and I thought I could modify [this line](https://github.com/evil-mad/axidraw/blob/90dcd9416d04ab487111eb6b0d2bb983d85e8183/cli/pyaxidraw/ebb_motion.py#L213) in the python CLI to pass through a different B pin, but I'm afraid that since QP doesn't report on anything but B1 (I think?), that would start to throw everything out of sync.

Am I completely crazy, or is there a decent way to drive two different pens? (Note: I never need to drive two pens at _once_.)

Sorry if this is all super basic, I don't do much hardware or low level software, so this is new for me!

Thanks!

oskay, 2020-02-24:
If you want to control separate pens separately, you'll need to do so with something like S2. The pen up / pen down functionality is only aware of one pen.

aarondfrancis, 2020-02-24:
Am I misunderstanding that the SP command is able to set the position on a different servo, with the third param?

`SP,value[,duration[,portBpin]]<CR>`

oskay, 2020-02-24:
So far as I know, that will work as it says there.

However, the ecosystem is generally built around the idea of there being one pen. There are not, for example, separate places to set multiple pen-up and pen-down positions, separate power controls for multiple pen-lift servos, multiple queries for different pen states, and the AxiDraw software knows only about pen-up and pen-down referring to a single pen, not multiple pens.

aarondfrancis, 2020-02-24:
Ok no worries. Thanks for the responses. We're currently just using a relay post-EBB board to do the switching, we'll keep doing that!

closed 2020-02-24 by aarondfrancis

oskay, 2020-02-24:
That's a clever solution! Can I ask what the application is? Knowing how things are being used can help guide design decisions in future firmware and hardware changes.

aarondfrancis, 2020-02-24:
Sure! We're actually writing on either side of the axidraw at different times, so we added a pen to the back side of the arm.

oskay, 2020-02-24:
Interesting! So, you are using one pen at a time, but switching which control output is being used-- correct?

aarondfrancis, 2020-02-24:
Yup! One pen at a time. When it comes time to switch, we send the active pen all the way to the top, and then send a signal from the Raspberry Pi to the relay to switch to the opposite pen, then we let the axidraw control it as if it's the standard pen.