evil-mad/EggBot #135: basestring does not exist in python3

jnweiger, 2020-04-24:
I've unpacked https://github.com/evil-mad/EggBot/releases/download/v2.8.1/eggbot-inkv1r2.zip
and find there, that eggbot_hatch.py references `basestring`
It should not.

https://docs.python.org/3.0/whatsnew/3.0.html#text-vs-data-instead-of-unicode-vs-8-bit
says
```
The builtin basestring abstract type was removed.
```

This was discussed in PR #134 before. Sorry, I still think I am right :-)

oskay, 2020-04-24:
I did not say, nor imply, that you were wrong.

oskay, 2020-04-24:
Is this producing an error for you in some instance, or is this just about code quality?

jnweiger, 2020-04-24:
I have not run the code. The python 2to3 tool spotted that. I'd asume it would not even compile as it is. Does it?

(No harm done. I'd be as happy when proven wrong!)

oskay, 2020-04-24:
I've never seen either the python2 or python3 version produce an error there.

closed 2020-12-24 by oskay

Stecki, 2021-04-03:
I also stumble upon this error message in inkscape, thus not being able to use hatch fills etc.; since not being proficient in python, I have no idea how to work around this issue.

oskay, 2021-04-03:
@Stecki What you are reporting is *different* from this closed issue.

If you are seeing an error, please open a new issue report, and include information about how you installed the software, which version of Inkscape, which OS, and what error message you are getting.