evil-mad/EggBot #139: EBB response pattterns

snoyer, 2020-06-30:
The EBB uses multiple different pattern to respond to various commands, for example:

`SP\r` -> `OK\r\n`, default ack: OK + newline
`V\r` -> `EBBv13_and_above EB Firmware Version 2.5.5\r\n`, data + newline
`QB\r` -> `1\r\nOK\r\n`, data + newline + OK + newline
`QM\r` -> `QM,0,0,0,0\n\r`, command + data +  different newline

This means that in order to communicate reliably with the EBB we need to know about every command and its expected response pattern.
While it is indeed necessary to know about every commands to meaningfully control the board, that should not be necessary at the communication level.

It would be better to use a single unambiguous response pattern and pick a single-character newline termination.
I think the `QM` response is the one to look at: command [+ optional data] + newline. The `OK` string becomes superfluous as the repeat of the command serves as an ack already. 
Appending an exclamation mark to the command could be used to indicate an error.

Using that pattern if we send, for example, `SP...\rSM...\rSM...\rQB\QM\r` we would get back `SP\rSM\rSM\rQB,1\rQM,0,0,0,0\r` 
instead of the current `OK\r\nOK\r\nOK\r\n1\r\nOK\r\nQM,0,0,0,0\n\r`.
In case of an error the response would look something like `SP\rSM\rSM!something went wrong\rQB,1\rQM,0,0,0,0\r`.

oskay, 2020-06-30:
I couldn't agree more that this would be an improvement. It's something that we've discussed internally many times.

Unfortunately, (1) the command set contains a lot of legacy "cruft" and (2) a whole lot of software built around the EBB now expects this. There isn't a graceful way of changing the command set like this without breaking most of the software that uses it.

snoyer, 2020-06-30:
I'm not surprised to hear that backward compatibility is a big issue here.
Would it be possible to implement this improved response mode and tuck it away behind a `CU` flag (similar to how `OK`s can be disabled) so that the old mode remains the default that legacy clients use, and new code has to request the new mode explicitly?

oskay, 2020-06-30:
That's... an excellent idea.

snoyer, 2020-06-30:
Awesome. I wouldn't mind looking into it myself if there is a way to emulate the firmware or compile a version that runs out of `stdin`/`stdout` rather than USB for testing in software; is that a possibility?

oskay, 2020-06-30:
We are in the middle of a major refactoring project right now; we'll look at whether and how this might fit into the current set of changes. It's probably not a great time to try and contribute a change like this, especially if there is a way for us to fit it in.

EmbeddedMan, 2020-06-30:
What we'd like to do here is make a clean break. The upcoming version 3.0.0 of the EBB firmware is the big refactor project oskay mentioned above. That version removes some old commands that are not used by the Inkscape extension, and adds several new ones, as well as bringing some interesting features along for the ride. In the process, we'll make a common return format, so that every command will reply with the command name, optional data, and a carriage return. All commands will be changed to be 2 characters in length. In addition, we'll make the error codes follow the same format, with a fixed length of 'optional data' (that being the error code string). All with the goal of making machine parsing of data coming back from an EBB simpler and cleaner.

This also means that the current mish-mash of return formats will not be supported going forward. 

The changes listed here will not be a big deal to implement for v3.0.0 - I've added them to my todo list.

Thanks for the great feedback, suggestions, and willingness to help out @snoyer .

snoyer, 2020-07-01:
That all sounds great, can't wait too see all of that.
This may be too late as the major refactoring is already underway but are you guys open to external feedback/requests and if so is there a place to discuss it without hijacking this issue?

Back on the topic of this one, are carriage returns "a thing" in the serial world and therefore preferable to line feeds?
`\n` may be nicer at a higher level, for example Python's [`readline`](https://docs.python.org/3/library/io.html#io.IOBase.readline), and therefore [PySerial's](https://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial.readline), assumes `b'\n'` for binary data.
That's obviously not critical as you could `read_until` with pyserial, open as text in python and specify the EOL, etc., but if backwards compatibility is out that kind of detail is worth considering.

EmbeddedMan, 2020-07-01:
I guess as far as I'm concerned, any combo of \n and \r is the same as any other from the EBB's perspective. It's just a PC program that's reading the lines, so let's make it as simple as possible to parse on the PC side. If just \n is better than \r or \r\n or \n\r, then let's go with it.

Also, if you do have feedback on the current EEB firmware, please feel free to add new issues. It's always good to hear what users would like.

snoyer, 2023-04-26:
Has there been any changes on this issue since `2.5.5` by any chance? Is this feature still in the cards?
I've got an itch to tinker with my axidraw again and I'd love to skip the friction of reading inconsistent EBB responses.

oskay, 2023-04-26:
This behavior has not been changed; we have still maintained the status quo for backwards compatibility reasons. We are still actively discussing (even in the past couple of weeks) if and when we will make a clean break requiring other software to update.

In case it is helpful, the AxiDraw software uses the following line, when sending a command `cmd`, to determine if an extra blank line should be read:
```
if cmd.split(",")[0].strip().lower() not in ["a", "i", "mr", "pi", "qm", "qg", "v"]:
    [read in extra blank/OK line]
```

If you are using Python, you can also use the [plotink library](https://pypi.org/project/plotink/) directly, calling `ebb_serial.query()` and `ebb_serial.command()` to wrap commands and queries, so that you do not have to remember the list.

snoyer, 2023-04-26:
Thank you @oskay for the swift update!

Thank you also for the Python pointers. I went and made [my own thing](https://github.com/snoyer/axidraw-control) back in the days but I'll check `plotlink` out and see what insights I can get.

I'd also volunteer for some beta testing/review/critique of the new stuff if that'd be of any help/interest.

EmbeddedMan, 2023-06-19:
I should have this implemented in the next couple days on a branch. We'll use CU,10,1 to turn on this 'harmonized line ending' mode. It will be off by default (to allow for backward compatibility). We will use '\n' as the line ending for every packet sent from EBB to PC over USB. It will be included in the upcoming v3.0.0 firmware release (not sure when that will be - hopefully pretty soon though).

EmbeddedMan, 2024-05-23:
EBB Firmware v3.0.0 and above have this feature. See ebb.html for complete documentation about it. Version 3.0.2 firmware was merged into master as of PR #232

closed 2024-05-23 by EmbeddedMan