evil-mad/EggBot #141: AttributeError: 'tuple' object has no attribute 'start'

lukehutch, 2020-07-07:
Latest Fedora Inkscape build (inkscape-1.0-2.fc32.x86_64) gives the following with the eggbot_hatch extension:

```
/home/luke/.local/lib/python3.8/site-packages/ink_extensions/cspsubdiv.py:12: DeprecationWarning: ffgeom.__init__ -> inkex.transforms.Segment(((x1, y1), (x2, y2)))
  s1 = Segment(p0,p3)
Traceback (most recent call last):
  File "eggbot_hatch.py", line 1736, in <module>
    e.affect()
  File "/home/luke/.local/lib/python3.8/site-packages/ink_extensions/inkex.py", line 285, in affect
    self.effect()
  File "eggbot_hatch.py", line 1215, in effect
    self.recursivelyTraverseSvg([self.selected[id_]], self.docTransform)
  File "eggbot_hatch.py", line 858, in recursivelyTraverseSvg
    self.addPathVertices(path_data, node, mat_new)
  File "eggbot_hatch.py", line 730, in addPathVertices
    subdivideCubicPath(sp, float(self.options.tolerance / 100))
  File "eggbot_hatch.py", line 575, in subdivideCubicPath
    if cspsubdiv.maxdist(b) > flat:
  File "/home/luke/.local/lib/python3.8/site-packages/ink_extensions/cspsubdiv.py", line 12, in maxdist
    s1 = Segment(p0,p3)
  File "/usr/share/inkscape/extensions/inkex/deprecated.py", line 257, in _inner
    return func(*args, **kwargs)
  File "/usr/share/inkscape/extensions/inkex/deprecated-simple/ffgeom.py", line 46, in __init__
    super(Segment, self).__init__((e0, e1))
  File "/usr/share/inkscape/extensions/inkex/transforms.py", line 723, in __init__
    start, end = other.start, other.end
AttributeError: 'tuple' object has no attribute 'start'
```

oskay, 2020-07-07:
Please use the Inkscape 1.0 version.

closed 2020-07-07 by oskay

lukehutch, 2020-07-07:
I am using the 1.0 version. The version number I gave is 1.0-2. "-2" is just the build number, not the software version number.

oskay, 2020-07-07:
Yes, exactly. Please use the build of this software _for Inkscape 1.0_, not the build for Inkscape 0.92.

lukehutch, 2020-07-07:
Ah, sorry, I misunderstood. So I assume that's the Inkscape_v1 branch? There are no instructions in `README.md` for this.

lukehutch, 2020-07-07:
I tried the Inkscape_v1 branch, and it gives the same error. The stack trace seems to indicate the problem is in ink_extensions? But there is no v1 branch of that project.

oskay, 2020-07-07:
If you want to use the version from the repository, please use Inkscape 0.92.

If you want to use Inkscape 1.0, please follow the instructions that we have provided for doing so.

We're not presently supporting any other paths besides these two.